ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #1: Configuration parameters read as ints although they might be floats

mleppanen, 2019-04-30:
For example startFreq expects an int on the cfg file, but it might be a float.

ivarvanwooning, 2019-05-31:
The solution is to change: `startFreq = int(splitWords[2])` into `startFreq = int(float(splitWords[2]))`.

The config file has some text, such as '62.00'. It can't directly convert this string to an integer. By first converting the string to a float and then converting the float to an int, the problem should be fixed. Don't forget to do this for the other variables:

`            startFreq = int(float(splitWords[2]))
            idleTime = int(splitWords[3])
            rampEndTime = float(splitWords[5])
            freqSlopeConst = float(splitWords[8])
            numAdcSamples = int(splitWords[10])`

ibaiGorordo, 2019-06-01:
I have corrected the function with Ivar's code. Thank you for letting me know.

closed 2019-06-01 by ibaiGorordo