ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #2: Association of points/tracking objects

ivarvanwooning, 2019-05-31:
If I understand correctly, this script visualises all the different detected points. The 1642 sensor also has a tracking algorithm to determine which points belong to which objects. Is there any way to receive this data as well?

ibaiGorordo, 2019-05-31:
Hello, you will have to do some changes to the code for that purpose. Depends on what demo you want to use. For example, for the people counting demo, you can find the information about the data packets at the end of the user guide.

[image]
First, the header seems to be different. Therefore my code (lines 169 to 186) needs to be changed.It should be something like:

```
        # Read the header
        magicNumber = byteBuffer[idX:idX+8]
        idX += 8
        version = format(np.matmul(byteBuffer[idX:idX+4],word),'x')
        idX += 4
        platform = format(np.matmul(byteBuffer[idX:idX+4],word),'x')
        idX += 4
        timeStamp = np.matmul(byteBuffer[idX:idX+4],word)
        idX += 4
        totalPacketLen = np.matmul(byteBuffer[idX:idX+4],word)
        idX += 4
        frameNumber = np.matmul(byteBuffer[idX:idX+4],word)
        idX += 4
        subFrameNumber = np.matmul(byteBuffer[idX:idX+4],word)
        idX += 4
        chirpMargin = np.matmul(byteBuffer[idX:idX+4],word)
        idX += 4
        frameMargin = np.matmul(byteBuffer[idX:idX+4],word)
        idX += 4
        uartSentTime = np.matmul(byteBuffer[idX:idX+4],word)
        idX += 4
        trackProcessTime= np.matmul(byteBuffer[idX:idX+4],word)
        idX += 4
        numTLVs = np.matmul(byteBuffer[idX:idX+2,word)
        idX += 2
        checksum = np.matmul(byteBuffer[idX:idX+2,word)
        idX += 2

```

Also, the tlv messages are completelly different, so the lines 201 - 255 need to be changed. Finally, the configuration file is going to be different so the parseConfigFile needs to be slightly changed. 

Tell me which demo you want to use and I can try to edit my code so it works for that demo, I think it can be helpful for other people also. Unfortunately, I do not have the AWR1642 board anymore so I will not be able to test it.

ivarvanwooning, 2019-06-02:
I'd like to use the people counting demo and obtain all the x and y coordinates for tracked objects. So for the parseConfigFile function, I will need to add some code for the SceneryParam, GatingParam, AllocationParam, TrackingCfg, etc.

So, to sum it up:
- Change the parseConfigFile method to also include the special configuration parameters for the people counting demo
- Change the header, like you explained in your comment
- Change the TLV message

This seems do-able! Let's see if it works :)

ibaiGorordo, 2019-06-13:
The new program for the people counting demo has been added. 

I close the issue.

closed 2019-06-13 by ibaiGorordo