ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #6: cant receive any data

TruongNguyen5395, 2019-09-17:
Hi I tried to run it, the graph appear but doesn't receive any data 
Im using SDK 02_00_00_04
CCS 9.1.0(url)

ibaiGorordo, 2019-09-18:
Hi,

I guess when you run the program the radar starts correctly ( LED DS3 turn on), right?

In that case, it may be some data parsing problem. Could you print the content of byteVec (write "print(byteVec" in line 119, and then send me the output?

Thank you,
Ibai

TruongNguyen5395, 2019-09-18:
Hi,

Thank you for replied my question, 
the result of byteVec:

[  2   1   4   3   6   5   8   7   4   0   0   2  66  22  10   0 247 154
  14 117 128   0   0   0 130  11   1   0   0   0   0   0  79   0   0   0
 183  80   0   0  34   0   0   0   9   6   0   0   1   0  96 100   7   0
   0   0  76   0   0   0  12   0   0   0  26  98   1 191 165 205  10  64
   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0 124 118
  51  65 118  54  55 189 161 181  60 190  57  54  55 189 154 254  50  66
 211 234 197 188 160 181  60 190 158 234 197 188 227 186  71  63 209   6
 227  64]
[  2   1   4   3   6   5   8   7   4   0   0   2  66  22  10   0  56  84
 167 117 152   0   0   0 131  11   1   0   0   0   0   0  79   0   0   0
 185  80   0   0   7   0   0   0 164   5   0   0   2   0 234 170   6   0
   0   0  24   0   0   0  65  40  14  64 127 146 106 190 157 249 165  61
  76  98 216  64   7   0   0   0  76   0   0   0  12   0   0   0  26  98
   1 191 165 205  10  64   0   0   0   0   0   0   0   0   0   0   0   0
   0   0   0   0 124 118  51  65 118  54  55 189 161 181  60 190  57  54
  55 189 154 254  50  66 211 234 197 188 160 181  60 190 158 234 197 188
 227 186  71  63   0   0 128  63]
[  2   1   4   3   6   5   8   7   4   0   0   2  66  22  10   0 247 199
  63 118 137   0   0   0 132  11   1   0   0   0   0   0  79   0   0   0
 140  80   0   0  40   0   0   0 177   6   0   0   2   0 160  53   7   0
   0   0  76   0   0   0  12   0   0   0  67 122   1 191   1 229  10  64
  86   2 171 187 144 244 173  60  18 246   7 189 135   4  13  62 100 245
  48  65 193 209  69 189   1   1 140 190 158 209  69 189 126 125  33  66
  81  58 119 188   1   1 140 190  79  58 119 188 214  72  53  63   0   0
 128  63   8   0   0   0   9   0   0   0  12]

i think that the data is correct. but still dont know where is the error

ibaiGorordo, 2019-09-18:
Hi,
 
Thank you for sending me the data. Are you sure you have the SDK demo? I am not sure but the data and the configuration file looks like the people counting demo. In that case, you can use the other python program ("peopleCountingDemo.py"). Otherwise try to flash the SDK demo and try again.

I hope it solves your problem. 

Best regards,
Ibai

TruongNguyen5395, 2019-09-18:
Oh...
so this code is use for the **xWR16xx mmWave Demo** right ? May be that is the reason i cant receive any things because of i flashed the **PeopleCountingLab**. and can i ask what is the industrial toolbox and the SDK you use to build the demo?

ibaiGorordo, 2019-09-19:
Hi,

If you want to use the people counting demo, use the "peopleCountingDemo.py" program. However, only the point cloud will appear. If you want to see the tracking information, you will have to use the **targetId** value.

The "readData_AWR1642.py" works only with the [mmWave SDK](http://www.ti.com/tool/MMWAVE-SDK) demo (for the SDK 3 and above, the data parsing I think it needs to be changed).

TruongNguyen5395, 2019-09-19:
Hi, 

Thank you for your advice, I did tried to flash the bin file in MMWave SDK demo 02_01_00_4,  and using your configuration 1642config to run readData also peopleCounting python file but still not receive any thing even those the LED DS3 is on.

the peopleCountingDemo.py return back this:

Traceback (most recent call last):
  File "peopleCountingDemo.py", line 378, in <module>
    dataOk = update()
  File "peopleCountingDemo.py", line 336, in update
    dataOk, frameNumber, targetObj = readAndParseData16xx(Dataport, configParameters)
  File "peopleCountingDemo.py", line 223, in readAndParseData16xx
    rangeVal = np.zeros(numInputPoints, dtype=np.float32)
ValueError: negative dimensions are not allowed

ibaiGorordo, 2019-09-20:
Hi,

The people counting program is not going to work with the SDK demo because the data message is different, that is why you get that error.

What I am not sure is why when you use the readData_AWR1642.py program, it stops after few seconds. 

- Is the data displayed on the graph before it stops? 

- Does any error appear when the data stops coming?

- Please make a print of byteVec and copy it here. If data from the radar stops coming, I am not sure what the problem might be. Maybe you can try to create a new configuration file using the demo visualizer app. If that configuration works well on the demo visualizer, it should work well also with the readData_AWR1642.py program.

TruongNguyen5395, 2019-09-20:
Hi, Ibai

You are correct after the Demo visualizer app to create new config file. the program readData_AWR1642.py work it turn out my old config was wrong. But now i want to focus on the peopleCountingDemo.py program after i flash the peoplecounting demo on the TI industrial tool and tried to load many config but it not working, it show that all the setting is loaded but the radar not work the LED DN3 is not turn on

i add an print(byteVec): 
dfeDataOutputMode 1
channelCfg 15 3 0
adcCfg 2 1
adcbufCfg 0 1 1 1 
profileCfg 0 77 30 7 62 0 0 24 1 128 2500 0 0 30
chirpCfg 0 0 0 0 0 0 0 1
chirpCfg 1 1 0 0 0 0 0 2
frameCfg 0 1 128 0 50 1 0
lowPower 0 1
guiMonitor 1 1 0 0
cfarCfg 6 4 4 4 4 16 16 4 4 50 62 0
doaCfg 600 1875 30 1
SceneryParam -6 6 0.1 15
GatingParam 4 3 2 0
StateParam 10 5 10 100 5
AllocationParam 450 0.01 25 1 2
VariationParam 0.289 0.289 1.0
trackingCfg 1 2 250 20 200 50 90
sensorStart
[]
[]
[]
[]
[]
[]
[]
[]
[]

TruongNguyen5395, 2019-09-21:
Hi,

 it is weird that i tried this configuration on people counting demo in matlab and it show the demo is success, i am received the data on the graph but with the python i receive nothing.

ibaiGorordo, 2019-09-22:
I think I have found the problem. The variable conversion from 4 byte to float32 was wrong. I have fixed it. I do not have the board so I cannot test if it works, but I have tested with your data and the results are the same as in MATLAB.

However, with the current version I think you will only see the position of the centroid for each target. If you want to visualize like in MATLAB you will have to edit the graph plotting (I do not know well about pyqtgraph so I cannot help you with that).

I hope with this solution your problem will be fixed and thank you for letting me know about what it was not working.

TruongNguyen5395, 2019-09-23:
Hi

Thank you, you helped me a lot. After i tried to run your new code the graph still not showing any data. However i tried Debug and i saw that the pointObj return the data dont know whether is it correct or not (**as you said that before this peopleCountingDemo.py just return the pointcloud data right?**). Maybe now i will try to take this value out and work with it.

ibaiGorordo, 2019-09-23:
Hi,

I am not sure what the problem might be. One thing I have realized is that in the data you sent me, the points were more than 2 meters away from the radar, but the scatter plot had a limit of 1.5 meters in the Y axis(I changed to 6 meters).

Also, I have changed what is being plotted. Before it was plotting the centroid where the target is (**targetObj["posX"]** and ["posY"]). Now, it is plotting the point cloud data that is stored in **pointObj**. 

I hope you are able to see something in the graph with these changes. If you cannot, print the data in targetObj and pointObj to see if they have strange values and let me know.

TruongNguyen5395, 2019-09-23:
Hi, Ibai

Yes it return the data on the graph but just about a second the there is no data return. i tried it with two others config and it show the same data.
So i print the data of numObj and targetObj then i see that the targetObj just return few times and it stop. this is the data i receive:

dfeDataOutputMode 1
channelCfg 15 3 0
adcCfg 2 1
adcbufCfg 0 1 1 1 
profileCfg 0 77 30 7 62 0 0 60 1 128 2500 0 0 30
chirpCfg 0 0 0 0 0 0 0 1
chirpCfg 1 1 0 0 0 0 0 2
frameCfg 0 1 128 0 50 1 0
lowPower 0 1
guiMonitor 1 1 0 0
cfarCfg 6 4 4 0 0 16 16 4 4 50 62 0
doaCfg 600 1875 30 1
SceneryParam -1.5 1 0.01 1.75
GatingParam 4 3 2 0
StateParam 2 5 10 200 5
AllocationParam 600 0.1 30 1 2
VariationParam 0.289 0.289 1.0
trackingCfg 1 2 250 20 200 50 90
sensorStart
{'numObj': 6, 'range': array([1.563452 , 1.563452 , 1.6123099, 1.563452 , 2.1497464, 2.1986043],
      dtype=float32), 'azimuth': array([0.1308997 , 0.09817477, 0.09817477, 0.06544985, 0.03272492,
       0.03272492], dtype=float32), 'doppler': array([ 0.08109608,  0.08109608, -0.08109608, -0.08109608, -0.08109608,
       -0.08109608], dtype=float32), 'snr': array([13.631447 , 20.507086 ,  6.7271996, 11.241763 ,  5.9119005,
        6.0045943], dtype=float32)}
{'targetId': array([0, 1], dtype=uint32), 'posX': array([ 0.12336747, -0.21300857], dtype=float32), 'posY': array([1.8189777, 0.6069746], dtype=float32), 'velX': array([0.07190235, 0.        ], dtype=float32), 'velY': array([0.00772061, 0.        ], dtype=float32), 'accX': array([0.22146273, 0.        ], dtype=float32), 'accY': array([-0.04043292,  0.        ], dtype=float32), 'EC': array([[[ 5.8318553 ,  9.708358  ],
        [ 1.8568538 , -0.24016786],
        [-0.041703  , -0.2707121 ]],

       [[ 1.8568538 , -0.24016792],
        [30.708668  ,  2.9909177 ],
        [-0.12572241, -0.12586609]],

       [[-0.04170299, -0.2707121 ],
        [-0.12572242, -0.12586609],
        [ 0.7724928 ,  0.62344056]]], dtype=float32), 'G': array([5.0987053, 1.       ], dtype=float32)}
{'numObj': 8, 'range': array([1.563452 , 1.563452 , 1.563452 , 1.563452 , 2.1497464, 2.1986043,
       2.1497464, 2.1986043], dtype=float32), 'azimuth': array([0.1308997 , 0.09817477, 0.06544985, 0.03272492, 0.03272492,
       0.03272492, 0.        , 0.        ], dtype=float32), 'doppler': array([ 0.16219215,  0.08109608,  0.08109608, -0.08109608, -0.08109608,
        0.08109608, -0.08109608,  0.08109608], dtype=float32), 'snr': array([12.954461 , 20.538279 , 19.037619 , 11.700562 ,  6.997252 ,
        7.1763287,  6.837403 ,  8.024067 ], dtype=float32)}
{'targetId': array([0, 1], dtype=uint32), 'posX': array([ 0.13324775, -0.21300857], dtype=float32), 'posY': array([1.8108511, 0.6069746], dtype=float32), 'velX': array([0.1274106, 0.       ], dtype=float32), 'velY': 
array([-0.0328301,  0.       ], dtype=float32), 'accX': array([0.3811107, 0.       ], dtype=float32), 'accY': array([-0.20218173,  0.        ], dtype=float32), 'EC': array([[[ 5.8725080e+00,  9.7083578e+00],
        [ 1.9275705e+00, -2.4016786e-01],
        [-2.2617150e-02, -2.7071211e-01]],

       [[ 1.9275705e+00, -2.4016792e-01],
        [ 3.1824722e+01,  2.9909177e+00],
        [-5.7050027e-02, -1.2586609e-01]],

       [[-2.2617146e-02, -2.7071211e-01],
        [-5.7050034e-02, -1.2586609e-01],
        [ 7.9852635e-01,  6.2344056e-01]]], dtype=float32), 'G': array([5.015355, 1.      ], dtype=float32)}
{'numObj': 9, 'range': array([1.563452 , 1.563452 , 1.6123099, 1.563452 , 1.563452 , 2.1497464,
       2.1986043, 2.1497464, 2.1986043], dtype=float32), 'azimuth': array([0.1308997 , 0.09817477, 0.09817477, 0.06544985, 0.03272492,
       0.03272492, 0.03272492, 0.        , 0.        ], dtype=float32), 'doppler': array([-0.08109608, -0.08109608, -0.08109608, -0.08109608,  0.08109608,
        0.08109608, -0.08109608,  0.08109608,  0.08109608], dtype=float32), 'snr': array([12.086596, 20.897211,  7.106682, 20.81749 , 12.58547 ,  6.836619,
        7.652691,  6.879309,  8.060236], dtype=float32)}
{'targetId': array([0, 1], dtype=uint32), 'posX': array([ 0.11030941, -0.21300857], dtype=float32), 'posY': array([1.8240954, 0.6069746], dtype=float32), 'velX': array([-0.07133469,  0.        ], dtype=float32), 'velY': array([0.0311905, 0.       ], dtype=float32), 'accX': array([-0.41359016,  0.        ], dtype=float32), 'accY': array([0.11838499, 0.        ], dtype=float32), 'EC': array([[[ 5.8347988e+00,  9.7083578e+00],     
        [ 1.8915771e+00, -2.4016786e-01],
        [-2.0220185e-02, -2.7071211e-01]],

       [[ 1.8915771e+00, -2.4016792e-01],
        [ 3.1811935e+01,  2.9909177e+00],
        [-8.5204795e-02, -1.2586609e-01]],

[... 30 lines cut ...]
        [-2.2974631e-02, -1.2586609e-01]],

       [[-1.6273391e-02, -2.7071211e-01],
        [-2.2974642e-02, -1.2586609e-01],
        [ 8.4204668e-01,  6.2344056e-01]]], dtype=float32), 'G': array([5.2718973, 1.       ], dtype=float32)}
{'numObj': 7, 'range': array([1.563452 , 1.563452 , 1.563452 , 1.563452 , 2.1986043, 2.1497464,
       2.1986043], dtype=float32), 'azimuth': array([0.1308997 , 0.09817477, 0.06544985, 0.03272492, 0.03272492,
       0.        , 0.        ], dtype=float32), 'doppler': array([-0.08109608,  0.08109608,  0.08109608,  0.08109608,  0.08109608,
        0.08109608,  0.08109608], dtype=float32), 'snr': array([12.217811 , 20.20881  , 21.710361 , 14.242106 ,  7.8724413,
        6.320089 ,  7.8431234], dtype=float32)}
[... cut ...]
{'numObj': 7, 'range': array([1.563452 , 1.563452 , 1.563452 , 1.563452 , 2.1986043, 2.1497464,
       2.1986043], dtype=float32), 'azimuth': array([0.1308997 , 0.09817477, 0.06544985, 0.03272492, 0.03272492,
       0.        , 0.        ], dtype=float32), 'doppler': array([-0.08109608,  0.08109608,  0.08109608,  0.08109608,  0.08109608,
        0.08109608,  0.08109608], dtype=float32), 'snr': array([12.217811 , 20.20881  , 21.710361 , 14.242106 ,  7.8724413,
        6.320089 ,  7.8431234], dtype=float32)}

TruongNguyen5395, 2019-09-23:
Hi, Ibai

Yes it return the data on the graph but just about a second the there is no data return. i tried it with two others config and it show the same data.
So i print the data of numObj and targetObj then i see that the targetObj just return few times and it stop. this is the data i receive:

{'numObj': 6, 'range': array([1.563452 , 1.563452 , 1.6123099, 1.563452 , 2.1497464, 2.1986043],
      dtype=float32), 'azimuth': array([0.1308997 , 0.09817477, 0.09817477, 0.06544985, 0.03272492,
       0.03272492], dtype=float32), 'doppler': array([ 0.08109608,  0.08109608, -0.08109608, -0.08109608, -0.08109608,
       -0.08109608], dtype=float32), 'snr': array([13.631447 , 20.507086 ,  6.7271996, 11.241763 ,  5.9119005,
        6.0045943], dtype=float32)}
{'targetId': array([0, 1], dtype=uint32), 'posX': array([ 0.12336747, -0.21300857], dtype=float32), 'posY': array([1.8189777, 0.6069746], dtype=float32), 'velX': array([0.07190235, 0.        ], dtype=float32), 'velY': array([0.00772061, 0.        ], dtype=float32), 'accX': array([0.22146273, 0.        ], dtype=float32), 'accY': array([-0.04043292,  0.        ], dtype=float32), 'EC': array([[[ 5.8318553 ,  9.708358  ],
        [ 1.8568538 , -0.24016786],
        [-0.041703  , -0.2707121 ]],
       [[ 1.8568538 , -0.24016792],
        [30.708668  ,  2.9909177 ],
        [-0.12572241, -0.12586609]],
       [[-0.04170299, -0.2707121 ],
        [-0.12572242, -0.12586609],
        [ 0.7724928 ,  0.62344056]]], dtype=float32), 'G': array([5.0987053, 1.       ], dtype=float32)}
{'numObj': 8, 'range': array([1.563452 , 1.563452 , 1.563452 , 1.563452 , 2.1497464, 2.1986043,
       2.1497464, 2.1986043], dtype=float32), 'azimuth': array([0.1308997 , 0.09817477, 0.06544985, 0.03272492, 0.03272492,
       0.03272492, 0.        , 0.        ], dtype=float32), 'doppler': array([ 0.16219215,  0.08109608,  0.08109608, -0.08109608, -0.08109608,
        0.08109608, -0.08109608,  0.08109608], dtype=float32), 'snr': array([12.954461 , 20.538279 , 19.037619 , 11.700562 ,  6.997252 ,
        7.1763287,  6.837403 ,  8.024067 ], dtype=float32)}
{'targetId': array([0, 1], dtype=uint32), 'posX': array([ 0.13324775, -0.21300857], dtype=float32), 'posY': array([1.8108511, 0.6069746], dtype=float32), 'velX': array([0.1274106, 0.       ], dtype=float32), 'velY': 
array([-0.0328301,  0.       ], dtype=float32), 'accX': array([0.3811107, 0.       ], dtype=float32), 'accY': array([-0.20218173,  0.        ], dtype=float32), 'EC': array([[[ 5.8725080e+00,  9.7083578e+00],
        [ 1.9275705e+00, -2.4016786e-01],
        [-2.2617150e-02, -2.7071211e-01]],
       [[ 1.9275705e+00, -2.4016792e-01],
        [ 3.1824722e+01,  2.9909177e+00],
        [-5.7050027e-02, -1.2586609e-01]],
       [[-2.2617146e-02, -2.7071211e-01],
        [-5.7050034e-02, -1.2586609e-01],
        [ 7.9852635e-01,  6.2344056e-01]]], dtype=float32), 'G': array([5.015355, 1.      ], dtype=float32)}
{'numObj': 9, 'range': array([1.563452 , 1.563452 , 1.6123099, 1.563452 , 1.563452 , 2.1497464,
       2.1986043, 2.1497464, 2.1986043], dtype=float32), 'azimuth': array([0.1308997 , 0.09817477, 0.09817477, 0.06544985, 0.03272492,
       0.03272492, 0.03272492, 0.        , 0.        ], dtype=float32), 'doppler': array([-0.08109608, -0.08109608, -0.08109608, -0.08109608,  0.08109608,
        0.08109608, -0.08109608,  0.08109608,  0.08109608], dtype=float32), 'snr': array([12.086596, 20.897211,  7.106682, 20.81749 , 12.58547 ,  6.836619,
        7.652691,  6.879309,  8.060236], dtype=float32)}
{'targetId': array([0, 1], dtype=uint32), 'posX': array([ 0.11030941, -0.21300857], dtype=float32), 'posY': array([1.8240954, 0.6069746], dtype=float32), 'velX': array([-0.07133469,  0.        ], dtype=float32), 'velY': array([0.0311905, 0.       ], dtype=float32), 'accX': array([-0.41359016,  0.        ], dtype=float32), 'accY': array([0.11838499, 0.        ], dtype=float32), 'EC': array([[[ 5.8347988e+00,  9.7083578e+00],     
        [ 1.8915771e+00, -2.4016786e-01],
        [-2.0220185e-02, -2.7071211e-01]],
       [[ 1.8915771e+00, -2.4016792e-01],
        [ 3.1811935e+01,  2.9909177e+00],
        [-8.5204795e-02, -1.2586609e-01]],
       [[-2.0220179e-02, -2.7071211e-01],
        [-8.5204795e-02, -1.2586609e-01],
        [ 8.0774629e-01,  6.2344056e-01]]], dtype=float32), 'G': array([5.143674, 1.      ], dtype=float32)}
{'numObj': 6, 'range': array([1.563452 , 1.563452 , 1.563452 , 2.1497464, 2.1986043, 2.1497464],
      dtype=float32), 'azimuth': array([0.1308997 , 0.09817477, 0.06544985, 0.03272492, 0.03272492,
       0.        ], dtype=float32), 'doppler': array([ 0.08109608, -0.08109608, -2.4328823 , -0.08109608,  0.08109608,
       -0.08109608], dtype=float32), 'snr': array([11.0769005, 18.220114 , 10.665834 ,  5.8635416,  6.3633327,
        6.322638 ], dtype=float32)}
{'targetId': array([0, 1], dtype=uint32), 'posX': array([ 0.1025631 , -0.21300857], dtype=float32), 'posY': array([1.8272665, 0.6069746], dtype=float32), 'velX': array([-0.12129411,  0.        ], dtype=float32), 'velY': array([0.01821502, 0.        ], dtype=float32), 'accX': array([-0.531608,  0.      ], dtype=float32), 'accY': array([-0.08549148,  0.        ], dtype=float32), 'EC': array([[[ 5.8456450e+00,  9.7083578e+00],     
        [ 1.9869671e+00, -2.4016786e-01],
        [-2.1119749e-02, -2.7071211e-01]],
       [[ 1.9869672e+00, -2.4016792e-01],
        [ 3.3127686e+01,  2.9909177e+00],
        [-3.8645670e-03, -1.2586609e-01]],
[... cut ...]
{'numObj': 7, 'range': array([1.563452 , 1.563452 , 1.563452 , 1.563452 , 2.1986043, 2.1497464,
       2.1986043], dtype=float32), 'azimuth': array([0.1308997 , 0.09817477, 0.06544985, 0.03272492, 0.03272492,
       0.        , 0.        ], dtype=float32), 'doppler': array([-0.08109608,  0.08109608,  0.08109608,  0.08109608,  0.08109608,
        0.08109608,  0.08109608], dtype=float32), 'snr': array([12.217811 , 20.20881  , 21.710361 , 14.242106 ,  7.8724413,
        6.320089 ,  7.8431234], dtype=float32)}

ibaiGorordo, 2019-09-23:
I don't know what is the problem. The data seems to be fine so I am not sure why it stops working.

When it stops working, try to see if there is data coming from the serial port (print on readBuffer). If there is nothing, try to increase the sleep time, the current setting (30 Hz) might be too fast but I am not sure that is the problem.

ibaiGorordo, 2019-09-23:
I've just realized about one possible reason. The program waits for the three types of TLV messages, and when it reads the last message (MMWDEMO_UART_MSG_TARGET_INDEX_2D) it activates the flag **dataOK**. However, if no tracking data is being received (as you sent in your previous message), maybe the **dataOK** flag is not being activated.

I have added the **dataOK** activation even if only the point cloud is being received. Just curious, try to print **numTLVs** to see how many TLV messages are being received in each package of data.

TruongNguyen5395, 2019-09-23:
I just do as you said and saw that when  3 TLV message sent the data will show in the graph but it just a few then it  send 1 TLV message

AllocationParam 600 0.1 30 1 2
VariationParam 0.289 0.289 1.0
trackingCfg 1 2 250 20 200 50 90
sensorStart
3
3
3
3
1
1
1
1
1
1
1
1
1
1
1
1
1
1

TruongNguyen5395, 2019-09-23:
Hi i tried to run the new code and i think that the data receive better now but it still remain in second and find no tlv were sent ( there is one time that the program receive the 3 tlv continously and the data show really well but just one time)

GatingParam 4 3 2 0
StateParam 2 5 10 200 5
AllocationParam 600 0.1 30 1 2
VariationParam 0.289 0.289 1.0
trackingCfg 1 2 250 20 200 50 90
sensorStart
3
3
3
3
1
1
1
1
1
1
1
1
1
1
1
1
1
0
0
0
0
0
0
0
0
0
0
0
0
0

TruongNguyen5395, 2019-09-23:
One more think that sometime i received this error:

AllocationParam 600 0.1 30 1 2
VariationParam 0.289 0.289 1.0
trackingCfg 1 2 250 20 200 50 90
sensorStart
3
3
2
Traceback (most recent call last):
  File "peopleCountingDemo.py", line 386, in <module>
    dataOk = update()
  File "peopleCountingDemo.py", line 347, in update
    x = -pointObj["range"]*np.sin(pointObj["azimuth"])
KeyError: 'range'

ibaiGorordo, 2019-09-23:
About the first problem, 0 TLV messages looks strange, try to print also the totalPacketLen to see the size of the package. Also increase the sleep time, maybe there is no enough data coming.

About the second problem, in that case only 2 TLV messages were received, so probably the point cloud TLV message was not received. To fix it, just remove the **dataOK** flag activation on the **MMWDEMO_UART_MSG_TARGET_INDEX_2D** message (line 312). By doing that, dataOK will only activate when there is point cloud data.

TruongNguyen5395, 2019-09-23:
Now im still dont know why the number of tlv after a while it reduce to 0 (some time remain at 1) this is an aout put that also print the totalPaketLen

cfarCfg 6 4 4 4 4 16 16 4 4 50 62 0
doaCfg 600 1875 30 1
SceneryParam -6 6 0.1 15
GatingParam 4 3 2 0
StateParam 10 5 10 100 5
AllocationParam 450 0.01 25 1 2
VariationParam 0.289 0.289 1.0
trackingCfg 1 2 250 20 200 50 90
sensorStart
108
1
92
1
108
1
108
1
140
1
108
1
92
1
140
1
204
1
140
1
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52
0
52

ibaiGorordo, 2019-09-23:
The packet length seems to be correct, only the 52 bytes from the header are being sent. I guess if nothing is detected by the radar no TLV are sent. However, if someone walks again inside the scene, some data should appear in the graph.

I am sorry, I do not know what might be the problem.

TruongNguyen5395, 2019-09-23:
Hi, Ibai

Thank you by the way, you helped me a lot. And i think that the problem could be in the config file so i will try to find out.

ibaiGorordo, 2019-09-23:
Good luck and let me know if you find some clue that can point to what is wrong.

TruongNguyen5395, 2019-09-24:
Hi, Ibai

Do you think that the problem could be in the configuration file because i checked many time and did not see any wrong from taking the bufffer to read the data in it ??

ibaiGorordo, 2019-09-24:
I finally realized the problem. When I wrote the program I was not expecting packets with no TLV messages. So, around line 316, the already processed data was removed from the buffer if the dataOK flag was activated. But in the case of no TLV messages, the dataOK flag is not activated so that data was not being removed from the buffer, and the program was analyzing that data continuously. 

Just remove the dataOK check in the if statement in line 316, like this:

```
# Remove already processed data
if idX > 0:
```

I think this will solve the problem.

TruongNguyen5395, 2019-09-25:
Hi, Ibai 

Yeah, i was thinking the same yesterday. Today when i run the code in spite of some time it crash due to the data streaming too fast but it work now, i receive the pointcloud also the targetID (delay few sec). So thank you so much, my project is to using this data and upload into Server so i will tried to optimize it. Once again, thank you so much!!!

ibaiGorordo, 2019-09-25:
Hi,

I am glad it finally worked. I will close this issue. If you have any feedback to improve the code I would love to hear.

Best regards,
Ibai

closed 2019-09-25 by ibaiGorordo