ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #7: Setting an array element with a sequence.

TruongNguyen5395, 2019-10-05:
Hi Ibai,

After using your peopleCountingDemo code for a while, then i found out out that there is two problem:
1. Firstly, when i run the program and it return the data for a while then it return an error:
             ValueError: setting an array element with a sequence
Then i found out that the problem is there could be an miss match with the initialize array so i changed: 
                azimuth = np.zeros(numInputPoints, dtype=np.float32)
                dopplerVal = np.zeros(numInputPoints, dtype=np.float32)
to
                azimuth = np.zeros(numInputPoints, dtype=object)
                dopplerVal = np.zeros(numInputPoints, dtype=object)
This will solve this problem.
2. Secondly, i some time receive the error

  File "peopleCountingDemo.py", line 215, in readAndParseData16xx
    tlv_type = np.matmul(byteBuffer[idX:idX + 4], word)
ValueError: matmul: Input operand 1 has a mismatch in its core dimension 0, with gufunc signature (n?,k),(k,m?)->(n?,m?) (size 4 is different from 3)

do you know how to solve this problem??

ibaiGorordo, 2019-10-07:
Hi, 

Sorry for the late response. I have corrected the first problem with your solution. I have no idea why it displays and error, thank you for letting me know.

About the second problem, I am also not sure. It seems that it is trying to multiply a vector of size 4 with a vector of size 3. My first guess was that **bytebuffer** was not full so only 3 bytes from **bytebuffer** were available. However, in the line 161 the length of **byteBuffer** is checked, so that problem should not happen.

Maybe the data is not being parsed correctly in those cases. Try to print the data in **bytebuffer** when those cases occur. Something like this
```
try:
        numTLVs = np.matmul(byteBuffer[idX:idX + 2], word)
        idX += 2
        checksum = np.matmul(byteBuffer[idX:idX + 2], word)
        idX += 2
except:
        print(byteBuffer)

```

TruongNguyen5395, 2019-10-22:
Hi Ibai,

I'am sorry for my late response too. But i found out the reason of that error which i think it come from the delay because when i adjust the time delay at 0.07 it lead to that error but when i adjust it back to 0.044 it doesn't receive that error but it change to  this error:

  File "peopleCountingDemo.py", line 188, in readAndParseData16xx
    byteBuffer[:byteBufferLength - startIdx[0]] = byteBuffer[startIdx[0]:byteBufferLength]                                                                                                                                                       
ValueError: could not broadcast input array from shape (0) into shape (31644)

So i think due to the program was too fast that lead to these errors.

ibaiGorordo, 2019-10-22:
Hello,

The problem was that I did not clear old data from the buffer. So it was making the data parser to think there was data even if the byteBufferLength was 0. I have corrected the code (actually another user suggested me this change a while ago but I forgot to fix it).

I have not tested it with this repository but I think it should work. Probably you can also reduce the time delay because that error should not appear again and you will see the data with less time delay.

Let me know if it worked.

TruongNguyen5395, 2019-10-22:
Hi, 

Thank for your reply. I tried your new code but the startIdx still not define so i tried to declare startIdx list before it then it return error that the list index out of range.

ibaiGorordo, 2019-10-22:
Hi,

Sorry, I copied the code in the wrong line, I think now it should work.

TruongNguyen5395, 2019-10-22:
Hi Ibai,

There is one problem that is the detObj at the 344 line i think this should be change to (targetObj["posX"]) > 0 and (targetObj["posY"]) > 0
right ? because the detObj still not define.

ibaiGorordo, 2019-10-22:
Yes, you are right. I forgot there was not detObj in this case. I used pointObj["range"] instead in case we want to visualize the raw point data even if there is no person detected.

TruongNguyen5395, 2019-10-22:
This is really strange that i change the code as you said but it report this error:

File "peopleCountingDemo2.py", line 494, in <module>
    dataOk = update()
  File "peopleCountingDemo2.py", line 431, in update
    if dataOk and len(pointObj["range"])>0:
KeyError: 'range'

I think it should be fine.

ibaiGorordo, 2019-10-22:
I am not sure, but you can delete that part (use only if dataOk:), I just put it in there so that the plot was updated when there is new data.

ibaiGorordo, 2019-10-22:
I also removed the second dataOK flag activation.Now the dataOk flag will only activate when the pointObj message is received.

TruongNguyen5395, 2019-10-22:
Hi Ibai,

Thank you so much, now the problem now solve and the program now run more stable. However after run for a while i still counter with the error: 

tlv_type = np.matmul(byteBuffer[idX:idX + 4], word)      (some time it the error happen in tlv_length)
ValueError: matmul: Input operand 1 has a mismatch in its core dimension 0, with gufunc signature (n?,k),(k,m?)->(n?,m?) (size 4 is different from 0)

I think that the problem could be come from the wrong TLV. so do you think can we add one function that ignore that wrong TLV ?

ibaiGorordo, 2019-10-22:
Thank you also, I wouldn't be able to correct the code wihtout your feedback.

I have added a try before the reading the tlv type and tlv length. It should skip the data if there is some error. However, I am not still not sure why that error occurs, I am probably missing something when parsing the people counting demo data.

TruongNguyen5395, 2019-10-23:
Hi Ibai,

i atry your code and the code will give error but i add an except and it work well now. 
   try: 
            # Check the header of the TLV message
                tlv_type = np.matmul(byteBuffer[idX:idX + 4], word) idX += 4
                tlv_length = np.matmul(byteBuffer[idX:idX + 4], word) idX += 4
  except:
                pass

thank you very much. Now the Programs run very stable now.

closed 2019-10-23 by TruongNguyen5395

fathia-ghribi, 2020-12-09:
hi guys !!
when i run this code `similarity = np.matmul(topic_vectors, word_vectors.T)
print(similarity`
i'm getting this error ValueError: matmul: Input operand 0 does not have enough dimensions (has 0, gufunc core with signature (n?,k),(k,m?)->(n?,m?) requires 1)
how to fix it ??

ibaiGorordo, 2020-12-11:
Hi @fathia-ghribi,

Probably is because of the shape of the input variables do not match, try print the shape of the variables to see if their shapes are correct (especially after you transpose the second variable).

However, that code is not from this repository and this issue has been close for over a year. Normally you will have more luck if you open a new issue in a more relevant repository for the code you are using.

fathia-ghribi, 2020-12-11:
> Salut @ fathia-ghribi ,
> 
> C'est probablement à cause de la forme des variables d'entrée ne correspondent pas, essayez d'imprimer la forme des variables pour voir si leurs formes sont correctes (surtout après avoir transposé la deuxième variable).
> 
> Cependant, ce code ne provient pas de ce référentiel et ce problème est résolu depuis plus d'un an. Normalement, vous aurez plus de chance si vous ouvrez un nouveau problème dans un référentiel plus pertinent pour le code que vous utilisez.

thnks so much for your answer

BhavyaSoni77, 2021-06-10:
> Hi @fathia-ghribi,
>
> Probably is because of the shape of the input variables do not match, try print the shape of the variables to see if their shapes are correct (especially after you transpose the [...]

Thankyou so much, this reply of yours helped me too!

adithi20, 2022-05-06:
[image]
can you please tell me where i am going wrong