ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #8: No output

lea92ch, 2019-10-17:
Hi,
I tried your code "Read-Data-Python", in my case, nothing is displayed on the graph window. I would like to know which .bin file you flashed to the radar?

Thank you

ibaiGorordo, 2019-10-18:
Hi,

In this case I made it for the mmWave SDK 2.01 demo. Which SDk are you using?

lea92ch, 2019-10-18:
I use mmWave SDK 3.03

ibaiGorordo, 2019-10-18:
Try using [my repository for the AWR1843](https://github.com/ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3-), change the configuration with your own configuration.

Let me know if it works. That program is still in development so it might give errors.

lea92ch, 2019-10-21:
With this repository and my own .conf I have only one point blinking on (0;0)

lea92ch, 2019-10-21:
This is with SDK 2.1
With SDK 3, nothing is displayed

ibaiGorordo, 2019-10-21:
When you use the SDK 2.1, does the sensor start measuring (the LDE DS3 should turn on). If it does, that mean that there is some problem while parsing the data. PLease make a print on byteBuffer and copy it here so I can see what could be the problem.

Thank you

cuca94, 2019-10-22:
I had the same problem, with my AWR1642 on SDK 3.03 i got no output. 
By using your AWR1843 repo, it works, but not all output data is parsed (z and velocity is not parsed).
This is the beginning of my byteBuffer output, i think the zeros in the 4th are parsed as z and velocity values..
btw: where do you got the information about the data stream from?

```
2   1   4   3   6   5   8   7   3   0   3   3 224   2   0   0  66  22
  10   0  22   0   0   0  37 164   2  77   6   0   0   0   4   0   0   0
   0   0   0   0   1   0   0   0  96   0   0   0 127  10  73 188  91 125
  43  61   0   0   0   0   0   0   0   0  31 243  33 189  54  23 151  60
   0   0   0   0   0   0   0   0  63 140 234  60 236 110 106  63   0   0
   0   0   0   0   0   0 211  82 207 190 164 151 141  63   0   0   0   0
   0   0   0   0 255  96 223  61  51 241 222  63   0   0   0   0   0   0
   0   0 223  33 240 190 210 129 232  63   0   0   0   0   0   0   0   0
   7   0   0   0  24   0   0   0 214   0 254   2 214   0 254   2 153   0
 247   2 205   0 233   2  23   1 240   2 183   0 204   2   2   0   0   0
   0   2   0   0  72  15  86  16  25  16 105  15 254  13 215  13 213  13
 100  12 239  10 173  11 155  12  67  13  35  14 111  13 179  12 186  11
 200  11  70  12 122  12 172  11 125  14  53  15  95  14  37  14 250  13
 197  14 184  15 215  15  37  15  22  14   5  13   1  12  70  11  68  11
 161  11 237  11  80  12  62  12 248  12 217  15  47  17   0  17 226  14
 252  14  81  14 102  14 146  14  45  14  27  13 170  13 170  13 222  12
 178  12 159  12 118  12 131  12 131  12 132  12 166  12  65  12  78  12
 184  12 209  12 228  11 215  11 117  12  54  12 194  11 103  11  37  11
 154  10 248  10  75  10  37  10 227  10  46  11  81  10  30  11 109  12
 121  12  47  12  37  11  12  11  33  11  95  11 241  10  33  11  80  11
 204  10 208  10  42  11 213  10  91  10 217   9  31  10 216  10 238  10
 115  10  58  10 126  10 182   9  84  10  16  10 133  10 153  10  29  10
 111  10 141  10  86  10   4  11 214  10 109  10 139   9 252   9  46  10
```

lea92ch, 2019-10-24:
I finally made it work using SDK 2.1. I don't know what the problem was before, it probably couldn't read data from Serial Port. I didn't have the problem since it started working so I can't tell you what the problem was so you could fix it. 
Thank you for your help

ibaiGorordo, 2019-10-27:
> I finally made it work using SDK 2.1. I don't know what the problem was before, it probably couldn't read data from Serial Port. I didn't have the problem since it started working so I can't tell you what the problem was so you could fix it.
> Thank you for your help

I am glad to hear it. Let me know if you find any problems again.

Thank you,
Ibai

ibaiGorordo, 2019-10-27:
> EDIT: Found the data structure of SDK2.1 and SDK3.3...
> EDIT2: ok, im just stupid...it gives zeros because of no moving objects 🤣
> 
> I had the same problem, with my AWR1642 on SDK 3.03 i got no output.
> By using your AWR1843 repo, it works, but the velocity is not sent. z is zero because the 1642 is not capable of calculating it, right?!
> This is the beginning of my byteBuffer output, the zeros in the 4th line are parsed as the velocity values..
> Any ideas why the velocity is not sent?
> 
> ```
>    2   1   4   3   6   5   8   7   3   0   3   3 224   2   0   0  66  22
>   10   0  22   0   0   0  37 164   2  77   6   0   0   0   4   0   0   0
>    0   0   0   0   1   0   0   0  96   0   0   0 127  10  73 188  91 125
>   43  61   0   0   0   0   0   0   0   0  31 243  33 189  54  23 151  60
>    0   0   0   0   0   0   0   0  63 140 234  60 236 110 106  63   0   0
>    0   0   0   0   0   0 211  82 207 190 164 151 141  63   0   0   0   0
>    0   0   0   0 255  96 223  61  51 241 222  63   0   0   0   0   0   0
>    0   0 223  33 240 190 210 129 232  63   0   0   0   0   0   0   0   0
>    7   0   0   0  24   0   0   0 214   0 254   2 214   0 254   2 153   0
>  247   2 205   0 233   2  23   1 240   2 183   0 204   2   2   0   0   0
>    0   2   0   0  72  15  86  16  25  16 105  15 254  13 215  13 213  13
> ```
Hi,
If I understand correctly, I think all your questions are solved. If you have any problems with the SDK 3, please write an issue in the other repository.

Thank you

closed 2019-10-27 by ibaiGorordo