ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #9: Potential to Plot 2D FFT

Lynnes001, 2019-11-21:
Hi,

I think you did a great work in parsing from TI's software. It will help me a lot. Thank you so much!

Just a question. I'd like to develop the 2D FFT plot on the basis of your code. The demo visualizer said it supports up to 4FPS realtime 2D FFT plot since limited by the bandwidth of USB. Can you give me some hint that which part I can start with extracting the 2D FFT plot? I really appreciate your open-source spirit and help!

Song

ibaiGorordo, 2019-11-21:
Hi, 

Thank you! I am glad to hear that. About the 2D FFT, you mean the range-azimuth heat map, right?

I tried to program the visualizer but I still have not male it work correctly. You can find the code in the heatMapDevelop branch of this repository.

I used the source code for the demo visualizer app to create the heat map visualizer. However, I probably made a mistake when comverting the code.

Let me know if you make it work,

Ibai

Lynnes001, 2019-11-22:
Glad to know you are working on this as well. I’ll check the branch. I’ll let you know if there Is any progress on my end.

Thanks again.
Song

ibaiGorordo, 2019-11-25:
Hi, 
I have just found another repository that also displays the heat map data. You may want to check it:
https://github.com/m6c7l/pymmw

Lynnes001, 2019-11-25:
Hi,

That should be really nice code from the screenshot. It's for 14xx boards. I'm not sure if it works on mine. Anyway, I'll have a try. Thanks!

closed 2021-12-24 by ibaiGorordo