ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #10: Enquiry

MiraKetbi, 2019-11-25:
We are three students from the American University of Sharjah. We are working on IWR1642 sensor to detect objects within 25m-30m. We used your code and it worked perfectly but we don't understand what is exactly giving us and what the angle is. We want our sensor to detect objects with 25-30m away and alert the users and the angle where the object is to program it later on with the GPS.

Can you please help us.

ibaiGorordo, 2019-11-25:
Hi,

I am glad to hear that the code works well. The _readAndParseData16xx_ function reads the data from the radar and returns a dictionary with the 2D position of the reflected points (X parallel to the radar and Y perpendicular to the radar). If you want to calculate the angle, you just have to do:
np.arctan2(detObj["x"], detObj["y"]) * 180 / np.pi

MiraKetbi, 2019-11-26:
Hey,
Thank you for your help, it helped a lot.
We want to add a plot of the angle vs the distance. We tried to do it but we couldn't.
Can you please help us

ibaiGorordo, 2019-11-26:
Hi,

What part were you able to do? And where did you get stuck?

MiraKetbi, 2019-12-01:
Hey, 
sorry for the late reply. 
we were able to find the angle and x and y coordinates but what we want is the **plot of the angle and distance**.. we couldn't do it

Also, the project basically we want our device to detect the objects within the range of 20-30m with the help of the GPS we will program both to work together.. We understand that the plot shows the Doppler velocity and it shows many points that don't make sense. What do those points represent? and how can we do the plot of angle vs. distance as in the IT Demo visualizer?

ibaiGorordo, 2019-12-03:
Hi,

The program shows the 2D position (horizontal plane) of the reflected points. It is similar to the demo visualizer demo. If you want to plot the angle range plot, just change the update function like this:

`
def update():
     
    dataOk = 0
    global detObj
    x = []
    y = []
      
    # Read and parse the received data
    dataOk, frameNumber, detObj = readAndParseData16xx(Dataport, configParameters)
    
    if dataOk and len(detObj["x"])>0:
        #print(detObj)
        angle = np.arctan2(detObj["x"], detObj["y"]) * 180 / np.pi
        range = np.sqrt(detObj["x"]**2 + detObj["y"]**2)
        
        s.setData(angle ,range)
        QtGui.QApplication.processEvents()
    
    return dataOk
`

You will also have to change the axis limits of the plot in the lines 307 and 308, write in **maxRange** the maximum distance you want to see:

```
p.setXRange(-90,90)
p.setYRange(0,**maxRange**)

```

Let me know if you have any other questions.

MiraKetbi, 2019-12-03:
Hey, 

Thank you so much for the code, it worked. 
The thing is the plot is giving us many points which don't make sense. 

Can you please help us in modify the code so we get reasonable points of objects around us?
Thank you again Ibai

MiraKetbi, 2019-12-05:
Hey, 
hope everything is well with you. 
I'm trying to save the plots as images but when I save it, it saves nothing (empty pictures)

In which loop should I put the saving function in?
Thank you in advance

ibaiGorordo, 2019-12-07:
Sorry for the late response,

About the points not making sense, if you want to improve the results you will need to develop some algorithm to filter noise points (for example DBSCAN for clustering points that are close together), but I cannot help you with that.

Regarding saving the data into an image, I am not sure how to do it, but you can try to copy the following in the **update()** function  after **_s.setData(x,y)_**

```
exporter = pg.exporters.ImageExporter(s.plotItem)
# set export parameters if needed
exporter.parameters()['width'] = 100   # (note this also affects height parameter)
# save to file
exporter.export('fileName.png')
```

closed 2020-01-06 by ibaiGorordo