ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #12: Range_Doppler Heat Map

harryD1, 2019-12-06:
Good work there IbaiGorordo. Can your matlab code display range-Doppler heat maps from the AWR1642?

ibaiGorordo, 2019-12-07:
I have eddited the **readAndParseData16XX.m** to also read the heatmap data. However, I took the code from the commented lines in the demo visualizer code so I am not sure if the code works correctly. If you want to visualize the heat map uncomment the lines 181 to 184 in the **readAndParseData16XX.m** function.

The problem is that I did not find any other demo that used the heat map from the mmwave sdk binary. 

If my code does not work, try running the occupancy detection demo in the automotive toolbox (you need to program the radar with its binary file).

Best regads,
Ibai

Kaczubie, 2019-12-29:
Hello,
I have written a Python code that does what you need, it is tested and works fine. I need only to add calculation of the y axis scale. 

```
import numpy as np

def calculate_range_doppler_heatmap(payload):

    rangeDoppler = list()
    for i in range(int(len(payload)/2)):
        rangeDoppler.append(payload[2*i] + payload[2*i + 1]*256)

    rangeDoppler = np.reshape(rangeDoppler, (config.numDopplerBins, config.numRangeBins),'F') #Fortran-like reshape
    rangeDoppler = np.append(rangeDoppler[int(len(rangeDoppler)/2):], rangeDoppler[:int(len(rangeDoppler)/2)], axis=0)

    range_matrix = np.multiply(np.arange(0, 257), config.rangeIdxToMeters)
    
    return rangeDoppler, range_matrix
```
[image]

Kaczubie, 2019-12-29:
It turned out that calculation of the y axis values is pretty straightforward.  
` y_scale = np.multiply(np.arange(-config.numDopplerBins/2 , config.numDopplerBins/2), config.dopplerResolutionMps)
`

`
self.dopplerResolutionMps =  3e8 /(2 * self.startFreq_actual * 1e9 * (self.idleTime + self.rampEndTime)*1e-6 * self.numDopplerBins * self.numTxAnt)
`

Best regards,
Kacper Zubiel

harryD1, 2019-12-31:
Okay, Thanks a lot I really appreciate. So how does the entire code looks
like?

ibaiGorordo, 2020-01-06:
Thank you @Kaczubie for your help. I have added the script to visualize the range-doppler heatmap with your code. I have tested it on the IWR6843AOP, but I think it should work also with the AWR1642. 

However, I have not added the cfg file, so you will have to create your own cfg file or modify the **guiMonitor** line in the current cfg file.

Let me know if you have any problems. The onl

ibaiGorordo, 2020-08-24:
Hi, 
I am sorry for the late response. Actually, I have not been able(/have the time and need) to get the Range-Doppler map even in Python. 

My recommendation is to search for a lab in the Industrial toolbox or Automotive Toolbox that has the Range-Doppler GUI in Matlab (I think there was none, but maybe I missed it). Otherwise, check the source code for the mmWave Demo Visualizer, there is MATLAB code commented for the processing of the Range-Doppler heat map.

Robiul-Islam-Minto, 2020-08-25:
Thank you very much.

Robiul-Islam-Minto, 2020-09-30:
Hi, 
I was trying to log data like mmwave demo visualizer but seems like my code doesn't save the whole data, instead, it saves single frame data. Have you tried to save the data? could you able to help me with this, below I have added my code  
-----------------------------------------------------------------------------------------------------------------------------------
for i = 1:100
numObj= detObj.numObj;
x = detObj.x
y = detObj.y
z = detObj.z
doppler = detObj.doppler
snr = detObj.snr
noise = detObj.noise
end
 save('DATA22.mat','numObj', 'x', 'y', 'z', 'doppler', 'snr', 'noise')

regards
Md Minto

ibaiGorordo, 2020-09-30:
Hi,

you just need to add the index to each variable:

```
for i = 1:100
     numObj(i)= detObj.numObj;
     x(i,:) = detObj.x;
     y(i,:) = detObj.y;
     z(i,:) = detObj.z;
     doppler(i,:) = detObj.doppler;
     snr(i,:) = detObj.snr;
     noise(i,:) = detObj.noise;
end
save('DATA22.mat','numObj', 'x', 'y', 'z', 'doppler', 'snr', 'noise')
```

 have not tested, so you might need to modify it (now it assumes that the detObj variables are row). Also, it is probably better if you initialize the matrices before the loop.

Ibai

Robiul-Islam-Minto, 2020-10-01:
Thank you, 
I was trying this way also but the code is unable to perform due to the left and right side have a different number of elements.
Another thing is, this code only showed the same data repeatedly, like this 
-------------------------------------------------------------------------------------------------------------------------------------
0.030348569	0.060635932	0.54572332	0.86857212	0.24876876	1.7811557	1.8713973	2.0849507	2.1539557	3.5987506	3.6868286
0.030348569	0.060635932	0.54572332	0.86857212	0.24876876	1.7811557	1.8713973	2.0849507	2.1539557	3.5987506	3.6868286
---------------------------------------------------------------------------------------------------------------------------------------

Regards 
Md Minto

ibaiGorordo, 2020-10-01:
True, I forgot about it, then just use a cell instead of a matrix:
`x{i} = detObj.x;`

Ibai

Robiul-Islam-Minto, 2020-10-01:
Thanks for the reply,
yes, I found the result but the problem is every cell has the same results. For example x{1,1} and x{1,2} and so on has the same value. 
-0.081761472	0.062683791	-0.59277070	-0.31069359	0.51782262	0.58595723	-0.11719144	-0.40063119	-0.70859939	1.0547230	7.6310711
these values are repeated for every x cell. What we expecting are different values. 

Regards
Md Minto

ibaiGorordo, 2020-10-01:
Hi, 

That might be another problem, can you send me your code? I will give a look when I have time.

Ibai

Robiul-Islam-Minto, 2020-10-01:
Thank you. I would be very grateful. 
I have mailed you the code. 

Regards
Md Minto

Robiul-Islam-Minto, 2020-10-04:
I think I have found where is the problem. It was inside the main loop. 

myInd = myInd + 1; 

Regards 
Md Minto

ibaiGorordo, 2020-10-05:
Hi, 

Sorry for not answering you earlier, yes, I think that is the problem.

Stan850, 2021-01-06:
Hi, has anyone been able to figure out why the rangeDoppler values in some frames are above 10000? Running the script with the rangeDoppler heat map logic added gives me values > 10000 for all frames (for rangeDoppler).

ibaiGorordo, 2021-01-08:
Hi,

To be honest I do not remember much about the script for the Range-Doppler (it has been exactly one year). Looking at the code I see that I also got that problem for some of the frames. I am not sure why, probably I missed some calculations.

I am not planning to work on the heatmap, so if you are interested in it I would suggest you to check this other repository that seems to have it working correctly:
https://github.com/m6c7l/pymmw

Ibai

Stan850, 2021-01-08:
Ok thanks Ibai, I will take a look!

closed 2021-12-24 by ibaiGorordo