ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #13: there is no replay and result in 2D scatter plot window.

gaowei-96, 2019-12-19:
hi,
When i run this code, it doesn’t work.
I use sdk 02.01, use your configfile.
In the console it only has these:
[image]

No other errors.
And the plot window is:
[image]
Can you help me? Thanks!

ibaiGorordo, 2019-12-20:
Hello,

Does your board start measuring when you send the config file? (the DS3 led should turn on).

Try to make a config file with the demo visualizer that works with your board, the config file might not work with your version.

Also, make sure the serial ports are the correct ones in your computer.

If the program does not run without errors, normally is either a problem with the config file or the serial ports.

Let me know how it goes.

Ibai

gaowei-96, 2019-12-20:
> Hello,
>
> Does your board start measuring when you send the config file? (the DS3 led should turn on).
>
> Try to make a config file with the demo visualizer that works with your board, the config file [...]

> Hello,
>
> Does your board start measuring when you send the config file? (the DS3 led should turn on).
>
> Try to make a config file with the demo visualizer that works with your board, the config file [...]

Hi,
I am sure that the serial ports are correct, the DS1 and DS2 led are turned on, but the DS3 led turn off.
I also think  the config file not correspond with my version. 
The SDK  is 02.01, when i try to use  the demo visualizer 2.1.0, it tell me the error that (red color):
[image]
i don't know why.
Hope for your replay,
Gao

ibaiGorordo, 2019-12-20:
Then you have two options, you can either program the board with the 2.1 SDK version, or you can use the [demo visualizer for the SDK 2.0](https://dev.ti.com/gallery/view/mmwave/mmWave_Demo_Visualizer/ver/2.0.0/) and generate a proper configuration file for your board.

gaowei-96, 2019-12-21:
Let me have a try. i'll tell you what happend.
Thanks.

closed 2020-01-06 by ibaiGorordo