ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #15: 2 D scatter plot

srivarshiniu1, 2020-02-18:
Hello!
Thank you so much for this code.
But here is the error i get while trying to run it.
**KeyError                                  Traceback (most recent call last)
<ipython-input-6-5e52007addc8> in <module>
     32     try:
     33         # Update the data and check if the data is okay
---> 34         dataOk = update()
     35 
     36         if dataOk:

<ipython-input-5-ce30366a076a> in update()
     15     if dataOk:
     16         print(targetObj)
---> 17         x = -targetObj["posX"]
     18         y = targetObj["posY"]
     19 

KeyError: 'posX'
**

I removed the comment on 
        x = -targetObj["posX"]
        y = targetObj["posY"]
as 
        x = -pointObj["range"]*np.sin(pointObj["azimuth"])
        y = pointObj["range"]*np.cos(pointObj["azimuth"])
gave an error :  " AttributeError: 'numpy.ndarray' object has no attribute 'sin'".

I'm a beginner and any help is apreciated!

ibaiGorordo, 2020-02-18:
Hello,

I have some questions:

- Are you using the default demo? Or did you program the board with the people counting demo?
- Does the board start measuring whe you run the program? (The led DS3 should turn on).

If you are using the default demo of the mmWave SDK, you should run the readData_AWR1642.py program. The peopleCountingDemo.py program is for the people counting demo.

Best regards,
Ibai

srivarshiniu1, 2020-02-18:
Hello!
Thanks for the fast response.
1. I have run the peoplecounting demo on the board.
2. The DS3 led does not turn on.
A plot opens but it is not responding.

I have also tried to run the readData program. It does not even open the
plot.
Please advise.

Srivarshini

ibaiGorordo, 2020-02-18:
Hello, 

I see. In the case of the people counting demo I am not sure if the DS3 turns on. Are you using Python 3? The numpy error might happen if you are using Python 2, in that case, try using Python 3.

For the first error, write print(targetObj) in the line 348 and tell me what it shows, like this :
```
dataOk, frameNumber, targetObj, pointObj = readAndParseData16xx(Dataport, configParameters)
print(targetObj)
```

Ibai

srivarshiniu1, 2020-02-18:
Hello!
I am using Python 3.

This is what happens when I print targetObj

dfeDataOutputMode 1
channelCfg 15 3 0
adcCfg 2 1
adcbufCfg 0 1 1 1
profileCfg 0 77 30 7 62 0 0 60 1 128 2500 0 0 30
chirpCfg 0 0 0 0 0 0 0 1
chirpCfg 1 1 0 0 0 0 0 2
frameCfg 0 1 128 0 50 1 0
lowPower 0 1
guiMonitor 1 1 0 0
cfarCfg 6 4 4 4 4 16 16 4 4 50 62 0
doaCfg 600 1875 30 1
SceneryParam -6 6 0.05 6
GatingParam 4 3 2 0
StateParam 10 5 10 100 5
AllocationParam 450 0.01 25 1 2
VariationParam 0.289 0.289 1.0
PointCloudEn 1
trackingCfg 1 2 250 20 200 50 90
sensorStart
{'targetId': array([12, 15], dtype=uint32), 'posX': array([
0.19347389, -0.34095234], dtype=float32), 'posY': array([2.0646744 ,
0.38558346], dtype=float32), 'velX': array([ 0.10368765, -0.17577091],
dtype=float32), 'velY': array([-0.03771946, -0.1050179 ],
dtype=float32), 'accX': array([ 0.29379326, -0.03419411],
dtype=float32), 'accY': array([-0.293203  ,  0.21192312],
dtype=float32), 'EC': array([[[ 1.0834372e+01,  1.1304497e+01],
        [ 5.6381203e-02, -7.6864734e-02],
        [-2.3492806e-01, -1.4206201e-01]],

       [[ 5.6381188e-02, -7.6864779e-02],
        [ 3.7184612e+01,  2.7410479e+00],
        [-2.3093631e-03,  4.3354448e-02]],

       [[-2.3492804e-01, -1.4206205e-01],
        [-2.3093726e-03,  4.3354467e-02],
        [ 7.3429060e-01,  7.9947144e-01]]], dtype=float32), 'G':
array([6.643481 , 2.8998523], dtype=float32)}
{'targetId': array([12, 15], dtype=uint32), 'posX': array([
0.19347389, -0.34095234], dtype=float32), 'posY': array([2.0646744 ,
0.38558346], dtype=float32), 'velX': array([ 0.10368765, -0.17577091],
dtype=float32), 'velY': array([-0.03771946, -0.1050179 ],
dtype=float32), 'accX': array([ 0.29379326, -0.03419411],
dtype=float32), 'accY': array([-0.293203  ,  0.21192312],
dtype=float32), 'EC': array([[[ 1.0834372e+01,  1.1304497e+01],
        [ 5.6381203e-02, -7.6864734e-02],
        [-2.3492806e-01, -1.4206201e-01]],

       [[ 5.6381188e-02, -7.6864779e-02],
        [ 3.7184612e+01,  2.7410479e+00],
        [-2.3093631e-03,  4.3354448e-02]],

       [[-2.3492804e-01, -1.4206205e-01],
        [-2.3093726e-03,  4.3354467e-02],
        [ 7.3429060e-01,  7.9947144e-01]]], dtype=float32), 'G':
array([6.643481 , 2.8998523], dtype=float32)}
{'targetId': array([12, 15], dtype=uint32), 'posX': array([
0.20054398, -0.3200538 ], dtype=float32), 'posY': array([2.0577135 ,
0.41546452], dtype=float32), 'velX': array([0.12829992, 0.07883841],
dtype=float32), 'velY': array([-0.08048633,  0.14931974],
dtype=float32), 'accX': array([0.32187694, 1.1216865 ],
dtype=float32), 'accY': array([-0.4259829 ,  0.97844565],
dtype=float32), 'EC': array([[[ 1.08639479e+01,  1.13736267e+01],
        [ 3.80516686e-02, -1.27354637e-01],
        [-2.13596746e-01, -1.03111148e-01]],
[... 124 lines cut ...]
        [ 6.76523149e-02, -9.64589342e-02],
        [-2.13729963e-01, -1.04173318e-01]],

       [[ 6.76523000e-02, -9.64589268e-02],
        [ 3.75653152e+01,  2.94211173e+00],
        [-6.24888111e-03, -4.10609618e-02]],

       [[-2.13729963e-01, -1.04173325e-01],
        [-6.24887785e-03, -4.10609394e-02],
        [ 7.54278362e-01,  8.40262175e-01]]], dtype=float32), 'G':
array([6.422044 , 2.9527338], dtype=float32)}

{}
{}

---------------------------------------------------------------------------KeyError
                                 Traceback (most recent call
last)<ipython-input-6-ef6dd554171f> in <module>     36     try:     37
        # Update the data and check if the data is okay---> 38
dataOk = update()     39      40         if dataOk:
<ipython-input-5-775845698863> in update()     16     if dataOk:
17         print(targetObj)---> 18         x = -targetObj["posX"]
19         y = targetObj["posY"]     20

KeyError: 'posX'

ibaiGorordo, 2020-02-19:
Hello,

That problem happened because no person was detected. I have modified the code to work either to show the detected target or to show the reflected points. I cannot test it, so please test the new code and let me know if it works.

About the numpy problem, I am not sure what might be the problem. You can try to print(pointObj) similarly to see what the problem might be.

Let me know how it goes.
Ibai

srivarshiniu1, 2020-02-19:
Hello!
Thank you!
The plot is showing up properly now.

My guess is that only one if these conditional statements are required.
    if targetDetected:
        #print(targetObj)
        x = -targetObj["posX"]
        y = targetObj["posY"]

    #if dataOk:
        #x = -pointObj["range"]*np.sin(pointObj["azimuth"])
        #y = pointObj["range"]*np.cos(pointObj["azimuth"])

If my guess is incorrect, please let me know.  For whatever reason I cannot
run the second statement.
Also, is there a way to reduce the sensitivity factor ?
Much Thanks!
Srivarshini

ibaiGorordo, 2020-02-22:
Sorry for the late response. 

In case you want to visualize the position of the person you can use  the if targetDetected statement. Otherwise, if you want to visualize the point cloud use the dataOk statement, but I am not sure why you get the numpy error. 

About the sensitivity, I have not used the people counitng demo, so I am not sure about it. You should check the documentation or ask in the Texas Instruments forum, they probably can help you better.

Ibai

srivarshiniu1, 2020-02-24:
T

srivarshiniu1, 2020-02-24:
Thank you !!
I shall look into the sensitivity part.

srivarshiniu1, 2020-02-28:
Hello!
I noticed that sometimes after sometime of working properly this error
comes up.
Any idea why and how to fix it?

ValueError                                Traceback (most recent call
last)<ipython-input-1-267e4deaf0f9> in <module>    427     try:    428
        # Update the data and check if the data is okay--> 429
dataOk = update()    430     431         if dataOk:
<ipython-input-1-267e4deaf0f9> in update()    355     356     # Read
and parse the received data--> 357     dataOk, targetDetected,
frameNumber, targetObj, pointObj = readAndParseData16xx(Dataport,
configParameters)    358     #change made    359     #prev_no=[0]
<ipython-input-1-267e4deaf0f9> in readAndParseData16xx(Dataport,
configParameters)    289                     accY[objectNum] =
byteBuffer[idX:idX + 4].view(dtype=np.float32)    290
   idX += 4--> 291                     EC[0, 0, objectNum] =
byteBuffer[idX:idX + 4].view(dtype=np.float32)    292
   idX += 4    293                     EC[0, 1, objectNum] =
byteBuffer[idX:idX + 4].view(dtype=np.float32)
ValueError: setting an array element with a sequence.

Thanks!

Srivarshini

ibaiGorordo, 2020-03-01:
Hello,

I am not sure what might be the problem. Try to run the program in interactive mode:
`python -i peopleCountingDemo.py`

And when the error occurs, write the following in the command line:
`print(byteBuffer)`
`print(idX)`
`print(len(byteBuffer))`
`print(tlv_length)`
`print(objectNum)`

Probably it is easier to see what the problem might be after analyzing those variables.

Best regards,
Ibai

closed 2020-05-11 by ibaiGorordo