ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #16: Sdk version

aditikarande, 2020-03-13:
Hi,
Is there a limitation to sdk version.
because i am using this code with my configuration file, and it dose not work.

ibaiGorordo, 2020-03-13:
Hi,

Yes, this repository is for the SDK 2. What SDK do you have in your board?

Best regards,
Ibai

aditikarande, 2020-03-13:
Thank you quickresponse.

I have SDK 3.3

ibaiGorordo, 2020-03-13:
I see. Then, try to use the code in [my other repository](https://github.com/ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3-). You may need to use your own config file, but it should work.

Let me know if you have any problems.

Ibai

aditikarande, 2020-03-13:
Hi,
there was a little confusion, i have board with sdk 2 version.
Now i am trying to read data for vital sign detection.
and i sent my configuration file, but its not working.

Aditi

ibaiGorordo, 2020-03-14:
I see. The vital sign demo has a different structure than the mmWave demo, so this program is not going to work. 

I can try to help you with the code. Which of the vital signs demos are your trying to run? And what data do you need?

Ibai

aditikarande, 2020-03-16:
Hi,
I am trying to run Driver Vital signs lab.
I tried to send the configure file to sensor, but its not working.
I do have configure file provided by the TI's lab.

LSS-David, 2020-03-17:
Hi.
I tried to same Driver Vital sign lab. 
but i faced same problem that above mentioned.
and data is not gathered ... mmwave studio or vital sign GUI are working.
pleases give me advice for deal with it... I have iwr1642boost(es:2.0)

ibaiGorordo, 2020-03-17:
Hi, 

The program is not working because the configuration file and the data that is being received is completelly different. I will try to update a program for the vital sign lab, but it might take time. 

I will post it here when I update the program.

Ibai

aditikarande, 2020-03-17:
Hi,

Yes, i agree.
but i have changed the configuration file , so it should be sent to the module right?
I didnt execute complete program, but i am just trying to send the file to module.

Aditi

ibaiGorordo, 2020-03-19:
I have added the program for the vital signs demo. I have not tested in a device so there might be some errors.

Also, it is still not complete. I have not implemented the code to decide which rate (FFT or peaks) to use.

Let me know if you have any issues with the code.

Ibai

aditikarande, 2020-03-19:
Thank you so much .

I will check and will get back to you asap.

Aditi

LSS-David, 2020-03-20:
Oh. thank you. 

Me also try it,

David

ibaiGorordo, 2020-03-23:
Hello,

Did any of you have problems with the code?

LSS-David, 2020-03-23:
Hello,

I changed sliceData > Dataport. 
and It works well. 

Thank you.

ibaiGorordo, 2020-03-23:
Hi,  

Thank you for letting me know. I will fix it now.

Ibai

aditikarande, 2020-03-23:
Hi,
All the things are good, but i am getting unboundlocal error for vitalSignsObj.
When the readandparse function is called in def update().
and in python3 does pyqtgraph work?

Aditi

aditikarande, 2020-03-23:
Hi,
Actually i forgot to instialize the vitalSignsObj, so my bad.
sorry for that.
now everything is working .
Thanks a lot for helping.
Aditi

ibaiGorordo, 2020-03-23:
Hello Aditi,

Thank you for your feedback, I have modified the initialization of vitalSignsObj.

I will close this issue. If you have more questions/problems with the vital signs demo, feel free to open a new issue.

Ibai

closed 2020-03-23 by ibaiGorordo