ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #17: regarding 2D scatter plot

jung-sang, 2020-05-11:
Hi ! 
im a student working on IWR1642boost and fortunately found your post on github !
Everything was going well, including installing and configuration etc also didnt see any errors
but
I dont get any result on 2D scatter plot screen...
So my assumption is... maybe its because my pi3 is not fast enough to run IWR1642boost in real time ?
So i wanna ask
Is pi3 okay to run the codes you wrote ?
also if its okay, then what do you think what my problem would be ,, ?
i loaded the peoplecounting demo on my board, so i works well on my PC
id appreciate if you advice me !
thank you for your time !

ibaiGorordo, 2020-05-11:
Hello,

Thank you for your message. I normally use the mmWave SDK demo in the Raspberry Pi Zero W, it it 
a different program but there should not be a problem with the Pi3.

Some things you can check are:
- Does the radar start measuring when you run the program?
- Did you use the same configuration on the PC and on the Pi? 
- Make sure the com port is correctly selected in your Pi, use "ls /dev/tty*" before and after you connect the radar to see the new ports. Then, copy those ports into the Python file. Normally, the configuration port number is smaller than the data port but if it does not work, try the number order.
- How do you run the program? There might be some program that might not show the plot.

Hope one of those things cna help you run the program. Let me know how it goes.

Best regards,
Ibai

jung-sang, 2020-05-11:
Hi,

it doesnt run on pc and my pi,
what i said was the board works well with TI gui demo, sorry for confusing you 

check the link below, i posted a screenshot.
https://imgur.com/gPXYfaD

as you see the photo, it doesnt show any errors also just stuck in 2d scatter screen without any results.

thanks again for replying me !

ibaiGorordo, 2020-05-11:
Hello,

Sorry, I have just realized that the program was using the wrong configuration file (it should use the configuration file for the people counting demo). I have corrected the program and uploaded the configuration file for the people counting demo.

You can also use your own configuration file (the one for the people counting demo) with the modified parameters to run the program.

Hope this solves your problem,
Ibai

jung-sang, 2020-05-11:
Thank you ! it sorted out the problem !

have a great day !

ibaiGorordo, 2020-05-11:
I am glad to hear that, I will close the issue then.

Let me know if you have any other problem.

Ibai

closed 2020-05-11 by ibaiGorordo