ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #18: regarding driver vital sign

jung-sang, 2020-05-12:
Hi again !

i could see some plots on people counting demo with configuration you edited last night,

but found the same issue on driver vital sign.

i loaded the same configuration file as TI one, 

and cant see any results like yesterday.

if you dont mind , would you take a look your code and tell me why it doesnt work ?

i guess the configuration might make problems like yesterday

best regards

ibaiGorordo, 2020-05-12:
Hi, 

Yes, you are right, the configuration file is missing. I will search for it and upload it. You can also find it in the vital signs demo from Texas Instruments.

Ibai

jung-sang, 2020-05-12:
Hi,

yeah its missing , so i copied it from Texas Instruments and pasted it to the same directory where the other codes are in, but doesnt work.

thank you

ibaiGorordo, 2020-05-12:
Hi, 

I have added the configuration files. Try with those configurations and let me know how it goes. 

Actually, I no longer have the board so I have not tested the code myself, so I am not sure what might be the problem.

Ibai

jung-sang, 2020-05-12:
Hi,
Sadly, it doesnt work. it shows nothing on 2d scatter plot screen.
thank you for all your amazing work.
i got a board like a week ago , so i was testing the board with a hope that it might work quite well on Pi.
Things im very interested about this board are people counting and heart rate detection.
again thank you very much!

ibaiGorordo, 2020-05-12:
Try to uncomment line 355 to print the vitalSignsObj variable. If nothing appears, try to print frameNumber around the line 256 to see if the data is being read correctly. 

Also, if you can test the vital signs demo from Texas instruments in your PC, and if it works try to use the same configuration with this program.

Let me know how it goes.

Ibai

jung-sang, 2020-05-12:
I sorted it out

it was my mistake,.,, lmao i didnt remove the jumper which  was on SOP2

thank you for everything !

ibaiGorordo, 2020-05-12:
I am glad to hear that.

Let me know if you have any other question/problem.

Ibai

closed 2020-05-12 by ibaiGorordo

Daya-AWR1642, 2022-04-22:
> I sorted it out
> 
> it was my mistake,.,, lmao i didnt remove the jumper which was on SOP2
> 
> thank you for everything !

Hai!!!
I too facing problem to run code 2d plot window pop up but it won't show anything simply rounds .
can you please help to fix it