ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #19: can't see the number of people detected

serendipitydt, 2020-05-18:
Hi,

I think you did a great work  from TI's software. It will help me a lot. Thank you so much!

Just a question. I want to use your code to realize all the functions of matlab code provided by Texas Instruments. Both of showing point cloud and showing how many people are there. This peopleCountingDemo.py  can only achieve show a point cloud,can't show how many people are there, and which point cloud is a person. Can you give me some advice to achieve that or another demo?  cause I am a new learner, it's really difficult for me,I really appreciate your open-source spirit and help!

Qsing

ibaiGorordo, 2020-05-19:
Hello Qsing,

Thank you for your message.

First of all, if you want to know the number of people, you can print the numTargetPoints variable. I have modified the code to print the number of detected targets if you uncomment the lines 353-357 and comment the lines 359-361, like this:
```
if targetDetected:
        #print(targetObj)
        print(targetObj["numTargets"])
        x = -targetObj["posX"]
        y = targetObj["posY"]
#if dataOk: 
        #x = -pointObj["range"]*np.sin(pointObj["azimuth"])
        #y = pointObj["range"]*np.cos(pointObj["azimuth"])
```
With this modifications, you will plot the position of the detected targets (people) instead of the raw point cloud and it will print the number of people in the console. If you want to continue plotting the raw point cloud data and print only the number of people, then you can add print(numTargetPoints) in line 261 and it will print the number of detected people.

I hope this answer your question. Also, if you check targetObj dictionary, it stores other information like the position of the people or their velocity.

Let me know if you have more questions,
Ibai

serendipitydt, 2020-05-20:
Hi! Bro
  It can really works！
   And i want to know that if there are any solution  to make a circle of the person detected instead of some point  cloud , like the  picture Ti's matlab code achieveed i send to you yesterday

ibaiGorordo, 2020-05-26:
Hello,
Sorry for the late response.

I have modified the code to plot both the radar point cloud (points) as weel as the position of the targets (square). To generate a plot like the one in Matlab with a bigger circle, you will have explore the  pyqtgrapgh to see if it is possible.

Also, regarding different colors for each person, you can check the last example in [this documentation](https://pyqtgraph.readthedocs.io/en/latest/plotting.html). In that case, you would have to use the targetObj ["targetId"] variable to know the id of each person detected.

Let me know if you have more questions,
Ibai

closed 2020-10-01 by ibaiGorordo