ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #20: Regarding the vital signs demo

Mixy1, 2020-05-23:
after around 5000-9000 frames the device keeps printing the same values. is this from the python script's end or the device's? and have you ever experienced this? I am running on Linux pop!_os 20.04 will try to run it on windows later.

ibaiGorordo, 2020-05-26:
Hello,

Actually I have not tested the vital signs demo myself, so I am not sure what might be the problem. You may want to modify the sleep time in line 403 (I am not sure if shorter or longer)  to see if the problem fixes.

Also, have you tried doing the same with the Texas Instruments? 

Best regards,
Ibai

Mixy1, 2020-05-27:
hey fixed it i just changed the buffer allocation as it was messing up the buffer. might do a pull request soon.

ibaiGorordo, 2020-05-27:
I see, thank you for letting me know. It might be a problem with the buffer cleaning or it might be fixed by reducing the sleep time.

Let me know if you find issues.

Ibai

closed 2020-10-01 by ibaiGorordo