ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #21: about ES2.0 EVM chip

sxontheway, 2020-05-28:
http://dev.ti.com/tirex/explore/node?node=ADSocWGX8FJCytCWREvHzw__VLyFKFf__LATEST

Hello, I am wondering is this script applicable for ES2.0 EVM? If not, what should can we modify? Thank  you.

ibaiGorordo, 2020-05-28:
Hello,
It depends in which demo version you have in your radar, which version do you have installed?

In the link you sent, they are using the mmWave SDK 3, in that case you should be using [my other repository for the SDK 3](https://github.com/ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3-).

Best regards,
Ibai

sxontheway, 2020-05-28:
Thank you so much! 
On my ES1 board, I use the .cfg and .bin of SDK 03.01, and everything works well. But on my new ES2 board (I haven't flushed .bin file yet), the code does not work. 
It seems that the problem is about the SDK version instead of ES1/ES2, since both ES1 and ES2 support SDK3. As long as the "xxx.bin" and the "xxx.cfg" use the same version of SDK, it will be fine?

ibaiGorordo, 2020-05-29:
Hello, 

Yes, the bin file version and the cfg version should be the same. However, if you are using the SDK 3 this repository should not work because the data parsing is different in version 3. So maybe the best option is to flush the SDK 3 into the new radar and use my other repository. 

Then, I think everything should work well. Let me know if it works well.

Ibai

SeungsooLeee, 2020-06-15:
@sxontheway 
Hello. sxontheway 
I met same issue. 1.0 worked well, but 2.0 doesn`t work. 
did you find correct flash version or config file? or other setting?

sxontheway, 2020-06-15:
> 
> 
> @sxontheway
> Hello. sxontheway
> I met same issue. 1.0 worked well, but 2.0 doesn`t work.
> did you find correct flash version or config file? or other setting?

Hi, I personally follow Ibai's steps and it works for me. First, flush the SDK 3.03, then use the parser in https://github.com/ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3- , just as what mentioned by Ibai.

ibaiGorordo, 2020-06-15:
Yes, probably using the SDK 3 as @sxontheway mentions will help you. Before that, could you tell me what SDK is installed in your board? It seems that the new board probably has the SDK 3 version, so probably I will create a new script in this repository for the ES2.0 board.

You can see the SDK version in the demo visualizer or if you use some Serial terminal (Putty, teraterm,...) connect to the configuration serial port and press the reset button. Then,  the board should print the SDK version installed.

Ibai

closed 2020-10-01 by ibaiGorordo