ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #22: Using your codes

SmellingSalt, 2020-05-30:
Hello,
**I would like to know which .bin file should be flashed onto the AWR1642** to be able to get the 2D plots working in your 'people counting demo' python code.

Also if **any information on other requirements** (like mmwave SDK version/CCS studio) is needed to run your demo, please do share them as well.
I am **not** using an the DCA1000EVM, and just have the AWR1642 board and my laptop, FYI.

I have run your python code but the 2D plot doesn't work and the 
dataOk = update()
line always returns dataOk as 0.

I am assuming it is because I have not flashed the correct .bin file.

**Could you also let me know as to which of the SOP2/SOP1/SOP0 switches are short while running your code?**

ibaiGorordo, 2020-05-30:
Hello,

In order to use the people counting demo you need to program your radar with the bin file for the people counting demo. For that, you need to download the Industrial Toolbox 4.1 (the latest version does not include the code for the AWR1642). Go to [this link](http://dev.ti.com/tirex/explore/node?node=AJoMGA2ID9pCPWEKPi16wg__VLyFKFf__4.1) and on the top right side press the download button. Once you have downloaded, open the labs folder -> people_counting -> 16xx_people_counting.

 I recommend you to check the user guide on the docs folder, it explains how to program the radar with the people countig demo usin the Uniflash software. Once you have installed the program correctly, you should be able to run the code in this repository.

Let me know how it goes.

Ibai

SmellingSalt, 2020-05-31:
Hello,
It worked, thank you so much!

closed 2020-05-31 by SmellingSalt