ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #23: configuration

FlorianMaier123, 2020-08-26:
Hi,
thanks for the code, it helped me a lot. My problem is, with your configuration file I cannot detect objects with a distance of more than about 3 meters. With the default settings in the mmwave demo visualizer, I can detect the object but when I copy this settings in the config-file for the python-programm, its not running. When I increase just the ramptime (and decrease the slope), I can detect the object but then I detect non-existing objects in front of the sensor too.
It would be nice if you can help me, Im new to the topic.

ibaiGorordo, 2020-08-26:
Hi,

Yes, the configuration is defined for short range. You should be able to use the same configuration file as in the demo visualizer. Just to make sure, which mmWaveSDK are you using? One possible issue could be this the program only works for the mmWave SDK 2 (I have other repositories for the other SDK versions), but since you are able to run with my configuration I guess that you are using the SDK2.

Did you also modify the range of the plot? Another option is that the objects are outside the range of the plot so that is why nothing is being shown.
```
p.setXRange(-0.5,0.5)
p.setYRange(0,1.5)
```

As for the noise near the sensor, do you get the same results in the demo visualizer? In that case, you might want to increase the CFAR threshold, or simply add some lines of codes to my program to remove the points that are too close to the radar.

FlorianMaier123, 2020-08-26:
Thanks for the respond,
yes i am using the 2.1 SDK. Its's not running if I copy the value for <numAdcSamples>, <digOutSampleRate> (ProfileCfg) but that shouldn't make a diffrence, right? Also, i cannot copy the value for CQRxSatMonitor and CQSigImgMonitor. I dont understand what these values are good for but it doesnt sound like its effecting the range. I already modified the plot. In the demo visualizer the object in front of the sensor is appearing and disappearing all the time but in the python programm i have always to objects about 10 cm in front of the sensor.

I added code to ignore the first 10 cm but it would be nice if I could oberve this area. The CFAR value didnt really help, since I either loose the object or get some new ones. I guess there is no other possibilty to not detect the objects in front of the sensor?
Thanks for your help.

FlorianMaier123, 2020-08-26:
just saw it deleted something in my text: Its's not running if I copy the value for numAdcSamples and digOutSampleRate, (ProfileCfg) but that shouldn't make a diffrence, right?

ibaiGorordo, 2020-08-27:
I cannot check the about the configuration, I will try to answer more in detail later. For the chirp configuration, you should probably ask in the Texas Instruments' forum, probably they can help you better than I do (I am not sure if it is the same but maybe this can help you https://e2e.ti.com/support/sensors/f/1023/t/810631).

As for the objects in front of the radar, I will have to look at it, but it should be the same results as in the demo visualizer, the code just parses the radar data and extracts the position of the points. One possible problem is that the program uses the values in the configuration to get the X, Y, Z of the points. So, if the configuration is not parsed correctly or in case the configuration is complex (I am not sure if there are any other ways to extract the 3D position in case of different configurations) there might be problem extracting the 3D position of the point cloud.

If you have the AWR1642 ES2.0, then I would recommend you to test the mmWave SDK 3, in that version the X,Y, Z is extracted directly from the radar raw data without using the values in the configuration.

ibaiGorordo, 2020-09-03:
Hi @FlorianMaier123,

were you able to get the configuration file to work correctly?

FlorianMaier123, 2020-09-03:
I stoped trying because I decided to get the AWR1843 (i want the z coordinate). When it arrives I will use your python script for it and see if it works. Im wondering if I can use it with the subframe configuration example I found in the doxygen. Im not sure if its possible to read each subframe or if I just get the detected objects for the whole frame.

Thanks for asking.

ibaiGorordo, 2020-09-03:
I have no experience with the advanced subframe configuration but even with the AWR1843 you might have to modify the code slightly. The program reads the subframe number, so then you can use that value to separate the data from the different subframes I guess. If you have any question with that please let me know and I will try to help you.

Ibai

closed 2020-09-30 by ibaiGorordo