ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #24: 2D scatter plot appears froze when running the code

sriza85, 2020-08-27:
Hi,

I tried to run the python file on my PC using readData_AWR1642.py but the 2D scatter plot will freeze every time. I attach the sceenshot of the 2D scatter plot. I am using config file for SDK 3.4 version based on the firmware from SDK 3.4.

[image]

ibaiGorordo, 2020-08-27:
Hi, 

this repository is for the mmWave SDK 2 (when I created the repository the AWR1642 used the SDK 2 by default, lately it has changed with the ES2.0). So, in your case you should use this repository instead:
https://github.com/ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3-

sriza85, 2020-08-28:
Hi Ibai, Thanks for the reply. I have done as what you suggested and it is still the same. I then flash the board with SDK2.1 and save the config file and use it. Then I use the previous repository but the same thing happened.

What I notice that I cannot even run the mmwave vizualizer demo. After I click sen config file to mmwave device, nothing happened.

What do you think the problem might come from?

Thank you for your help.

ibaiGorordo, 2020-08-28:
Hi, 
I am not sure what might be the problem, my first guess is that the configuration does not match with the firmware version, but if you tried with different versions, then probably is another problem. Maybe try to check that the jumpers are correctly selected. 

However, if the demo visualizer does not work correctly, then this demo is not going to work. Maybe you should ask in Texas Instruments' forum about it. Once you have it working on the demo visualizer, it should also work with this program.

Sorry,
Ibai

ibaiGorordo, 2020-09-03:
Hi,

were you able to fix the problem?

ibaiGorordo, 2020-09-04:
Hi, 
In that case, I think the best option would be to ask about it in the Texas Instruments' forum. 

But just to make sure, 

- What SDK have you flashed into the board? 
- Maybe try to flash the board with another SDK version.
- Make sure that the SOP jumpers are correctly selected in functional mode.
- Try to revert the selected COM ports (COM4 for data and COM3 for cfg)

sriza85, 2020-09-04:
Hi,

I have been doing all the suggestions above but still it does not work. I will raise this issue in TI forum.
Hopefully once this issue is sorted out, I can use your code since I want to use the radar with Rasberry PI.

Thanks again for your advice.

closed 2020-09-04 by sriza85