ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #25: configuration file

akbkon, 2020-09-14:
Hello,
thanks for your code! Would you please tell me the meaning of these parameters and the values after them in the configuration file(1642config.cfg)? Because I may have to modify some values according to my needs.  I'm a new learner and I'm confused.  

I really appreciate your help!

ibaiGorordo, 2020-09-15:
Hello,

The information about the parameters is defined in the User guide of the mmWave SDK: https://software-dl.ti.com/ra-processors/esd/MMWAVE-SDK/latest/exports/mmwave_sdk_user_guide.pdf 

You can modify the parameters using the mmwave sdk demo visualizer: https://dev.ti.com/gallery/view/mmwave/mmWave_Demo_Visualizer/ver/3.4.0/

Regards,
Ibai

akbkon, 2020-09-15:
Many thanks for your respond!

ibaiGorordo, 2020-09-16:
I am glad to hear that.

Let me know if you ave any other questions/comments.

closed 2020-09-16 by ibaiGorordo