ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #27: ValueError: matmul: Input operand 1 has a mismatch in its core dimension 0, with gufunc signature (n?,k),(k,m?)->(n?,m?) (size 2 is different from 0)

vinit-katariya, 2020-12-09:
I am using the  readData_AWR1642.py just to read the data from the sensor and getting this error for line 218 peakVal = np.zeros(tlv_numObj,dtype = 'int16'). I have checked the packet structure and it looks fine.

Do you know what can be the problem?

ibaiGorordo, 2020-12-11:
Hi, 

What is the output when you try to print tlv_numObj? 

Something similar happened in #7 but I am not sure what might be the cause. Try modifying the delay value. If it does not work you can try to add an exception as mentioned in the linked issue.

vinit-katariya, 2020-12-14:
Thank you for the reply. I will try that.

vinit-katariya, 2020-12-15:
It's working now, error is gone. It was an issue with the binary file. I was using an older version.

closed 2020-12-15 by vinit-katariya