ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #28: 数据采集的配置问题

chunguangqu, 2020-12-10:
您好，我问一下，采用您的代码实现数据的实时采集，雷达设备需要作出哪些设置么？因为我运行您的代码并没有运行成功，在117行的Dataport.in_waiting的值一直0，感谢您的回复，谢谢~

ibaiGorordo, 2020-12-11:
Hi, 

My guess is that the configuration file you are using does not match with the SDK version inside the radar. Are you using the SDK 3? 

If you are try to run the code using my other repository: https://github.com/ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3-

The problem is that the new version of the AWR1642 is shipped with the SDK 3, but when made this program it was not.

Hopefully that can fix your problem.

walidbrahim, 2021-08-09:
Hello 
I am having the same problem using AWR1443Boost board (ES3) with SDK 2.1, the value is always 0. I tried setting the config values using mmWave Visualizer but it did not solve the problem (it works in the visualizer).
 I also want to ask about the "magicword" variable in your code, does it depend on the board used?
Thank you for the work you've done here.

ibaiGorordo, 2021-08-09:
Hi Walid,
the fact that there is no data available in the data serial port means that the radar is not sending any data. So, it is either that the ports numbers are not correctly configured or that the configuration sent to the radar is not correct.

In that regard, 
- Did you save the configuration from the mmWave Visualizer and tested with this code? 
- Did you use the same serial ports in both cases?
- Also, just to make sure, in the demo visualizer, did you select also SDK 2,1?

The magic word is common for all boards, and it is used for detecting the beginning of the data message. Anyway, since you are not getting any data, it would not affect in this case.

Ibai

walidbrahim, 2021-08-10:
Hi 
Thanks a lot for your feedback. effectively, I needed to import the configuration from the visualizer into the .cfg file.

ibaiGorordo, 2021-08-10:
Thank you for letting me know, can you copy the configuration here? I want to check what might be the problem.

Thank you

walidbrahim, 2021-08-10:
Yes sure. I used this configuration (I am still new to the TI boards features and how to control them though):

sensorStop
flushCfg
dfeDataOutputMode 1
channelCfg 3 1 0
adcCfg 2 1
adcbufCfg 0 1 0 1
profileCfg 0 77 761 7 20 0 0 100 1 64 5333 0 0 30
chirpCfg 0 0 0 0 0 0 0 1
chirpCfg 1 1 0 0 0 0 0 0
frameCfg 0 0 64 0 100 1 0
lowPower 0 1
guiMonitor 1 1 0 0 0 1
cfarCfg 0 2 8 4 3 0 1280
peakGrouping 1 1 1 1 56
multiObjBeamForming 1 0.5
clutterRemoval 0
calibDcRangeSig 0 -5 8 256
compRangeBiasAndRxChanPhase 0.0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0
measureRangeBiasAndRxChanPhase 0 1.5 0.2
CQRxSatMonitor 0 3 4 37 0
CQSigImgMonitor 0 31 4
analogMonitor 1 1
sensorStart

I am working on detecting vital signs of a sleeping target. Is the AWR1443Boost a good device to start with. If you have any advice I will be grateful.
Thanks in advance.

ibaiGorordo, 2021-08-10:
Thank you! 

A good start for the vital sign is the demo in the Industrial toolbox. In Python, I have this code where it reads the vital signs, but it is not completed, so you might have to do some changes to parse the data.
https://github.com/ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2/tree/master/Driver%20vital%20signs%20demo

walidbrahim, 2021-08-10:
Thanks for the link and the advice.

closed 2021-08-10 by ibaiGorordo