ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #29: Is MATLAB Software needed

guruvishnuvardan, 2020-12-10:
Hi,

Can you please let me know, if MATLAB Software is needed to run this python demo on AWR1642.

Thanks
Guru

ibaiGorordo, 2020-12-11:
Hi Guru,

No, no MATLAB is needed for the Python scripts in this repository. You only need python and some of the libraries mentioned in the Readme. If you want to run the MATLAB scripts inside the "MATLAB code" folder then MATLAB is necessary. But for the rest of the scripts you can make it work without it.

Regards,
Ibai

guruvishnuvardan, 2020-12-11:
Hi Ibai,

Good Day!

Thanks for the response.

I am planning to use an Ubuntu 18.04 PC to connect AWR1642, Is there any specific thing,  I need to take care?
Can you please correct me, if my below points are wrong

1) Post Connection to PC and Setup

2) range-dopplerHeatmap_AWR1642.py or readData_AWR1642.py are used to get Range and Velocity of the Obstacles

3) Can you please let me know, if there is any specific reason to change the config file name?, I understand serial port will vary from PC to PC.

Note - Please bare with some of my questions, As I have just started with AWR1642 data extraction.

Thanks
Guru

ibaiGorordo, 2020-12-13:
Hi, 
if you want the point cloud data, then use the readData_AWR1642.py script.

As for the configuration, you can create your own custom configuration using mmWave Demo Visualizer. Particularly, in the case of the heatmap script and the normal point cloud script, first of all, the configuration file defines which type of data will the radar send to the computer: either 3d point cloud data, heatmap data, or both among other plot options.

You can also configure other things like the framerate (for example the heatmap can only run at a maximum of 5 fps due to the high amount of data), maximum range, range resolution. So my recommendation is to first play with different values in the demo visualizer, and once you have a configuration that works for your application, then use this script.

guruvishnuvardan, 2020-12-13:
Hi Ibai,

Thanks a ton.

I will work on the above mentioned points and get back to you for any clarification.

Regards
Guru

closed 2020-12-13 by guruvishnuvardan