ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #30: AWR1642 - Raspberry Pi 4(Raspbian OS)

guruvishnuvardan, 2021-01-18:
Hi Ibai,

Good Day!

Can you please clarify me the following:

1) If I can use Raspberry Pi 4 & Raspbian OS for working with AWR1642-Read-Data-Python-MMWAVE-SDK-2?

2) Can I also install AWR1642 SDK on Raspbian OS?

3) Is response of Raspberry Pi good(for real-time)?

Thanks
Guru

ibaiGorordo, 2021-01-20:
Hi Guru,

> If I can use Raspberry Pi 4 & Raspbian OS for working with AWR1642-Read-Data-Python-MMWAVE-SDK-2?
Yes

> Can I also install AWR1642 SDK on Raspbian OS?

I am not sure, but you do not need to install it in the Raspberry Pi. Actually, you only need it if you want to reprogram the firmware on your board. In that case, you can upload the firmware of the radar from your laptop and only use the Raspberry Pi for the measurement.

> Is response of Raspberry Pi good(for real-time)?

It depends on what you want to do with it, but I have tested this repository with a Raspberry Pi 3 and it worked fine. So, the Pi4 should be fine.

Ibai

guruvishnuvardan, 2021-01-20:
Thanks Ibai

closed 2021-01-20 by guruvishnuvardan