ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #34: Differntiating Horizantal Plane and Elevation Plane through Azimuth Plane

guruvishnuvardan, 2021-04-18:
Hello Ibai,

Good Day!

I am using AWR1642BOOST EVM to R&D ADAS applications, I see AWR1642 also picks few objects(like over head telephone cables, over head internet cables) on the elevation plane, I had requested TI Forum to help me differntiate the Elevation Plane and Horizantal Plane of detection, they had mentioned the following:

"The AWR1642BOOST EVM has all of the virtual antennas on the same plane. As a result, detections can only be differentiated along the aziumth plane. The angular resolution will be a function of the number of TX and RX antennas that are enabled in your particular configuration."

Can you please let me know, if its possible to differntiate using your code through Azimuth Plane.

Thanks
Guru

ibaiGorordo, 2021-04-18:
Hi Guru,

It is not a software problem, a radar that was all the antennas in the same plane (in this case, in the horizontal plane), cannot differentiate objects in the vertical plane. Check the angle of arrival part in this document to learn more about it:
https://www.ti.com/lit/spyy005

If you want to differentiate also in the veritcal plane you will have to change to the AWR1843, AWR1642BOOST-ODS or any of the other radars that do not have antennas in one plane.

guruvishnuvardan, 2021-04-19:
Hi Ibai,

Thanks for the Response.

My requirement is, I do not want the vertical plane (even in AWR1843), I am only looking for detecting objects on the road, Can you please let me know, if it's possible to only detect Obstacles on Road using AWR1843.

Thanks
Guru

ibaiGorordo, 2021-04-23:
Hi,

What you want is not to know the height of objects, but you want to only get the infromation of objects that are on the road (i.e.  have a low height)?

In that case you can filter the points based on their height, and then ignore the z value when you represent the point cloud (like in my AWR1843 repository).

guruvishnuvardan, 2021-04-23:
Thanks Ibai :)

closed 2021-04-23 by guruvishnuvardan