ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #35: PyQt5 install error at Raspberry Pi 4

superBaik, 2021-04-26:
Hello, 
First of all, thanks for your codes.

I got the error about setting PyQt5 at my raspberry pi 4 module.
As you mentioned, you've already tested at window and raspberry pi OS, I m just wondering if you didn't encounter any error regarding PyQt5 setting. 

The error is as below. 

Try. 
pip install PyQt5

-> No any response, just pending 

and main issue is 

Traceback (most recent call last):
  File "ppCnt1.py", line 4, in <module>
    import pyqtgraph as pg
  File "/usr/local/lib/python3.7/dist-packages/pyqtgraph/__init__.py", line 13, in <module>
    from .Qt import QtGui, mkQApp
  File "/usr/local/lib/python3.7/dist-packages/pyqtgraph/Qt.py", line 137, in <module>
    from PyQt5 import QtGui, QtCore, QtWidgets, sip, uic
ImportError: /usr/lib/python3/dist-packages/PyQt5/sip.so: undefined symbol: _Py_ZeroStruct

Could I get any reply ? 

Thanks

ibaiGorordo, 2021-04-27:
Hi,

You should probably ask that question in the pyqtgraph repository, I have not installed pyqtgraph in the reaspberry pi in a while. But it is possible to run pyqtgraph in a raspberry pi.

superBaik, 2021-04-27:
Thanks for your reply. 

Firstly, I could clear the above problem, just using python command not python3. 
I think  pip and pip3 installed was mixed.

I would ask one more thing. 
I will try to use IWR6843 EVM board using 1642 python people counting code. 
The big difference was the cfg file, so cfg file for 6843 has to be applied. 
but the source code, specially the part of readAndParseData function and logic has to be changed. 

Though I have yet to analyze the TLV data format from rardar sensor with cfg file, 
Can I ask to any advice to revise the code for customizing the 6843 EVM board ? 

thanks

ibaiGorordo, 2021-04-27:
Use my other repository instead:
https://github.com/ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3-

The 6843 probably comes with the sdk3, so that repository should work fine with the 6843.

Ibai

superBaik, 2021-04-27:
Thanks,  

I will try to do it as you mentioned. 

By the way, Can I ask you which TI mmWave example is applied to your code with ARW1843 ?
I mean, as you know there are a lot of examples related to AWR1843 examples, and also TLV messages and header also a little bit different. 

Could you please let me know the related TI example code applied to AWR1843 with SDK-3 ? 

Thanks again !!!

ibaiGorordo, 2021-04-27:
All the main codes in these repositories are for the default demo that comes with the radar, the only difference is that the default demo data has changed over the years, so the repository I mentioned should support the default demo that comes with the 6843.

closed 2021-12-15 by ibaiGorordo