ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #36: Question about range-doppler heatmap

chenxin926, 2021-04-26:
Hi my friend
Thank you for uploading the code, it is very useful
There is something wrong with the Doppler heat map when I run the code. Could you please advise me if there is something wrong with the.cfg file I configured and I can only use if np.max(RangeDoppler) >The picture will not appear until 30000 is changed to 50000, otherwise the picture will not be available. Could you please tell me the reason? Thank  you!
[image]
[image]
[image]
[image]
[image]

ibaiGorordo, 2021-04-27:
Hi, if that configuration works well with the demo visualizer, it is probably a problem with the implementation. Actually that if statement should not be there but I put it there becaue I was not sure why the range heatmap data was bad in some cases (it looks like in your case it is that way for all frames).

I am not sure if it will make any difference, but if you can update the firmware to the SDK3, then try my other repository (https://github.com/ibaiGorordo/AWR1843-Read-Data-Python-MMWAVE-SDK-3-), which is the version I tested for the heatmap plot.

Otherwise, you might want to check at the code in this repository as it seems that they are able to plot the heatmap correctly (I am budy with other projects now, so I cannot try to fix this implementation).

chenxin926, 2021-04-28:
Thank you for your reply. I will try to change your 1843 code to 1642 code, because the number of antennas is different. Thanks again

closed 2021-12-24 by ibaiGorordo