ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #40: Question about EVM flash file

ChrisBaxter8040, 2021-09-14:
I have a question regarding the use of other TI lab demos with this python program. I am using the IWR 1642BOOST from TI and It has the Obstacle detection binary flashed onto the EVM. Is it possible to use this python program to graph the output?

ibaiGorordo, 2021-09-15:
It depends on whether the data is transmitted in the same format as in the out of the box demo. In the case of the obstacle detection, the format is slightly different (image below), so you would have to modify the code in this repository a bit to work with that demo
[image]
Source: https://dev.ti.com/tirex/explore/node?node=AGQLWBtluPOb4IBBpo4JYw__AocYeEd__LATEST

ChrisBaxter8040, 2021-09-15:
Thank you for your reply. I will attempt the modification and provide my findings here

ibaiGorordo, 2021-09-16:
Good luck! It should not be that hard, but let me know if you have any questions.

ChrisBaxter8040, 2021-10-07:
Thank you for your help, I was able to parse the packet data via the use of the struct library in python.

closed 2021-12-15 by ibaiGorordo