ibaiGorordo/AWR1642-Read-Data-Python-MMWAVE-SDK-2 #43: DriverVitalSignLab

MariadeCarlos, 2022-01-17:
Kaixo Ibai! Eskerrik asko zure kodigoa zabaltzeagatik. 

I am trying to work with your code but I do not receive any data from your radar after using the cfg file for the vital sign sensing, and I can therefore not see anything in the GUI (It does not even open) 

I would appreciate it if you could help me in any way. 

I have flashed the device (AWR1642) with the SDK 2.01

ibaiGorordo, 2022-01-19:
Kaixo, eskerrik asko!

It has been a while since I used so I might be wrong. For the vital sign demo you will have to flash the binary that is included in the vital sign lab of the [automotive toolbox](https://dev.ti.com/tirex/explore/node?node=AFeCagqbt7.hu2lHOXE4eA__AocYeEd__LATEST) (the one in the industrial toolbox might work but I am not sure).

However, I am not sure if this code will work with the latest version or if something has changed lately.

MariadeCarlos, 2022-01-21:
Yes that was it! 
I tried it on automotive toolbox 2.5 and it works 
Thanks a lot

closed 2022-01-21 by MariadeCarlos

Daya-AWR1642, 2022-04-22:
> Yes that was it! I tried it on automotive toolbox 2.5 and it works Thanks a lot

Hi!!
I'm also facing same problem can you please help me to resolve it

mjk1376, 2023-03-29:
hello can anyone help me solve same problem. i can't find file in that link